libremesh/lime-packages #652: ubis-lime-metris get_gateway does not work with babel without ipv6

spiccinini, 2019-11-25:
Currently. as far as I understand, the code get's the gateway using a filter that only get's the gateway if the route if an ipv6: https://github.com/libremesh/lime-packages/blob/fa205c64c8630d07cfeacdf27173a68e3dd02a2c/packages/ubus-lime-metrics/files/usr/libexec/rpcd/lime-metrics#L110
It should work also when the gateway doesn't have an ipv6 address.

ilario, 2020-09-22:
The incriminated line moved here:
https://github.com/libremesh/lime-packages/blob/3883b01915b4e646d685812736c4e425b4ef1d21/packages/ubus-lime-metrics/files/usr/sbin/last_internet#L25

I didn't test, but looking at it seems that now what is needed is that the gateway direction starts with `10.`.
Seems that a network changing the default direction to something not starting with `10.` (e.g. `192.`) will not work here.

spiccinini, 2020-09-23:
Indeed, also 10.x may be the ip of a host outside the network (it happened to me a couple of times) so the gateway is outside the network.

ilario, 2020-09-23:
Rrrright!! So what do you suggest to do in order to get the real gateway? Going back to babeld dump?

spiccinini, 2020-09-23:
I am not sure. Maybe I would use the information of the "own" network segment (like 10.13.0.0/16) and then use mtr to find when it leaves this network segment. Maybe using luci.ip that I think can be used to ask an IP address is inside or not a network.

ilario, 2020-11-10:
What about trying to get `/cgi-bin/hostname` from each of the IPs, in order, until one fails?
The last one to give a meaningful answer is the LibreMesh gateway node.
If none succeeded, the node itself is the gateway node.
This requires the fix of #803

ilario, 2020-11-11:
Another alternative: check, in order, if the routers have open 6696/udp (babeld port).
But I have no idea how this can be done in a shell script.
On the router it is open, it can be checked with `netstat -tulpn`.
Weird thing: using nmap on my laptop (nmap -sU -p 6696 10.13.0.1), that port looks closed, both with or without the OpenWrt firewall (FW3) package installed.

ilario, 2020-11-11:
> What about trying to get `/cgi-bin/hostname` from each of the IPs, in order, until one fails?

Maybe better, let's publish a static file, something like `/www/libremesh_check.txt` with a content like "LibreMesh", and let's check all the IPs for the content of this file, until one fails the check.

spiccinini, 2020-11-11:
The Gateway should be the last node of the local mesh network, right? LibreMesh may be used in nodes that are non local to the network but are in the path. If we want to get the last node of our network thenwe should check the ip and the netmask, right?

ilario, 2020-11-11:
> The Gateway should be the last node of the local mesh network, right?

Whoooopppss I did not expect that, I thought that it was the node connected to the internet exit (the latest LibreMesh node before the modem or a non-LibreMesh router).
@gmarcos87 @germanferrero @nicopace can you clarify this please?

ilario, 2020-11-12:
> Weird thing: using nmap on my laptop (nmap -sU -p 6696 10.13.0.1), that port looks closed, both with or without the OpenWrt firewall (FW3) package installed.

I just realised that I didn't setup an interface with VLAN before attempting this, so it was normal that it failed. I will try again.

> The Gateway should be the last node of the local mesh network, right?

In this case, instead of looking for the content of a static file, we can expose the `ap_name` option in `/www/cgi-bin/ap_name` and then compare the content of this file on the other nodes with the content on the very same node.
The working principle is the same as the proposed check for `/www/libremesh_check.txt` but it will distinguish where the LibreMesh L2 local network finishes (the only supported way to create a border is to have two networks with different `ap_name`s). And do not need any complex IP check (so it does not pull any library from LuCI).

ilario, 2020-12-01:
> > Weird thing: using nmap on my laptop (nmap -sU -p 6696 10.13.0.1), that port looks closed, both with or without the OpenWrt firewall (FW3) package installed.
> 
> I just realised that I didn't setup an interface with VLAN before attempting this, so it was normal that it failed. I will try again.

Tried again, it works.

> In this case, instead of looking for the content of a static file, we can expose the `ap_name` option in `/www/cgi-bin/ap_name` and then compare the content of this file on the other nodes with the [...]

Can we decide on this? I am inclined to think that what makes more sense is until the last LibreMesh router before non-LibreMesh ones (regardless of local or non-local mesh).

ilario, 2023-03-18:
@selankon am I wrong or were you working on something related?

ilario, 2024-11-10:
> The Gateway should be the last node of the local mesh network, right? LibreMesh may be used in nodes that are non local to the network but are in the path. If we want to get the last node of our network thenwe should check the ip and the netmask, right?

This sounds very good.
We can make a filter based on this code:
https://unix.stackexchange.com/a/258926