libremesh/lime-packages #659: New network name created via FBW is forced to lowercase but default is mixed case

ilario, 2019-12-01:
FirstBootWizard forces the users to use lowercase names for the new networks (see https://github.com/libremesh/lime-app/commit/73d464e) but the default "LibreMesh.org" is not lowercase.

The lowercase was introduced as a bug fix ([mentioned here](https://github.com/libremesh/lime-app/commit/df460c8a65fc7034f3eb87ff6b3263cbcfa80185#diff-4ac32a78649ca5bdd8e0ba38b7006a1eR10)) but the bug is not specified, maybe is for avoiding confusion?

We should change the default "LibreMesh.org" to "libremesh.org".
A (maybe small but surely annoying for someone) drawback is that the `10.13.x.x` network domain would change to `10.249.x.x`

ilario, 2020-08-26:
Fixed with https://github.com/libremesh/lime-app/pull/268

ilario, 2021-01-21:
Maybe this is not 100% fixed, see https://github.com/libremesh/lime-app/issues/288

germanferrero, 2021-01-22:
Oh, no my fault, we do want hostnames to be lower case forced. I though we didn't. Network names are note forced to be upper-case. Thanks @ilario

ilario, 2021-01-22:
Mh...
Let's keep the "hostname" and "network name" issues clearly separated please, I am a bit confused now.

Problem 1:
Network name defaults to "LibreMesh" (in lime-system's lime-defaults) but the fbw (and lime-app?) forces it to be lowercase. Is this correct?

Problem 2:
Hostname defaults to LiMe-%M4%M5%M6 but fbw (and lime-app?) forces it to lower case. Is this correct?

germanferrero, 2021-01-22:
> Problem 1:
> Network name defaults to "LibreMesh" (in lime-system's lime-defaults) but the fbw (and lime-app?) forces it to be lowercase. Is this correct?

Not anymore, this was fixed by https://github.com/libremesh/lime-app/pull/268, now network name can use uppercase.

Problem 2:
Hostname defaults to LiMe-%M4%M5%M6 but fbw (and lime-app?) forces it to lower case. Is this correct?

Yes, we have closed https://github.com/libremesh/lime-app/issues/210 to soon.
@spiccinini said there:
> We should allow uppercase network names. Hostnames, in my opinion, in lowercase. @germanferrero ?

I didn't answer then, now I do: I think we should allow uppercase on hostnames too. It gives more freedom to the users and it is harmless. DNS resolution is case insensitive.

spiccinini, 2021-02-04:
> > We should allow uppercase network names. Hostnames, in my opinion, in lowercase. @germanferrero ?
> 
> I didn't answer then, now I do: I think we should allow uppercase on hostnames too. It gives more freedom to the users and it is harmless. DNS resolution is case insensitive.

I think it is better to use all lowercase hostnames, and change the default to lime-%M4%M5%M6, because there is a lot of broken software in the world that does not have case insensitivity support. 
I know that enforcing softly the all lowercase hostname is not a proper fix, and bugs may pile under the carpet (for example storing hostnames in hash maps without prior enforcing lowercase, etc).
In the other hand I think it is more confusing to have mixed case hostnames.

ilario, 2023-03-01:
So it would be enough to change this line:

https://github.com/libremesh/lime-packages/blob/78162b4d773c9e8d01f63801ce3464ea84ac89b6/packages/lime-system/files/etc/config/lime-defaults#L9

ilario, 2024-11-09:
Two networks with different ieee80211s names like LiMe-DDEEFF and lime-DDEEFF will not communicate between them, right?
So, if we really want to make this change we should make it during a big update where the back compatibility is broken anyway.

But personally I think that it is not worth, I would just leave that as it is...

ilario, 2024-11-10:
Two networks with different ieee80211s names like LiMe-DDEEFF and lime-DDEEFF will not communicate between them, right?
So, if we really want to make this change we should make it during a big update where the back compatibility is broken anyway.

Between the 2020.1 and the 2024.1 release there has been maany changes, but I am not sure they made the two release incompatible...

Anyway, personally I think that it is not worth to fix, I would just leave that as it is. Closing. FBW developers, please assess whether this is still needed and feel free to re-open if you consider so. @spiccinini @G10h4ck @selankon @gmarcos87

closed 2024-11-10 by ilario