libremesh/lime-packages #660: FBW: when creating a new network warn the client that has to connect to a new AP

ilario, 2019-12-01:
When creating a new network with FirstBootWizard, the AP name changes and if the user was connected via wireless will wait forever on a "Checking connection" page.

The situation:
I was connected with my laptop to a freshly flashed router, via wireless
I created a new network using the FBW

The problem:
the "Checking connection" page was making me wait forever

The solution:
I suppose that the only possible solution (I'm not sure...) is to include in the "Checking connection" page a message suggesting: "If you're connected via wireless, connect to the $new_essid_name access point"

germanferrero, 2020-10-30:
This issue has been addressed some time ago.

[image]

Can we close this issue?

ilario, 2020-11-04:
I checked, it's fixed. Thanks!

closed 2020-11-04 by ilario