libremesh/lime-packages #661: FBW scans for new networks only on 5 GHz radios

ilario, 2019-12-01:
FBW function to join an existing network does not work with 2.4 GHz only devices.
It will make the user wait and then return just a "no network found" message.

This is likely the cause of #589

The related code is:
https://github.com/libremesh/lime-packages/blob/3235c25a4bea489d54986e32f21ab220c7684580/packages/first-boot-wizard/files/usr/lib/lua/firstbootwizard.lua#L63-L71

A possible improvement would be to use 2.4 GHz radios only once the 5 GHz radios didn't find anything.

spiccinini, 2020-06-04:
Maybe we can filter out radios that do not have a mesh interface (so if 2.4GHz radios do not have a mesh interfaces then they are not used), what do you think?

ilario, 2020-06-04:
That would do!

ilario, 2022-03-22:
> I tested this yesterday with LROS 1.5, only using 2.4. One network found the other. Might be solved.

The causing code didn't change, it should not be fixed...
What do you mean with "only using 2.4"?

ilario, 2022-12-16:
@selankon let's continue here the discussion started on https://github.com/libremesh/lime-app/issues/324#issuecomment-1354477609 :
there are places in `firstbootwizard.lua` where the 5 GHz interfaces get listed, like here:
https://github.com/libremesh/lime-packages/blob/be6319b2b723c653c7c2e49b738b9c1504b1a707/packages/first-boot-wizard/files/usr/lib/lua/firstbootwizard.lua#L85

we could add a check and in case this list is empty do something else with the available interfaces.

We should address the following cases:

- [x] at least one 5 GHz and one 2.4 GHz (e.g. LibreRouter, TP-Link WDR3600...)
- [ ] no interface at 2.4 GHz (e.g. Ubiquiti NanoStation M5, TP-Link CPE 510)
- [ ] no interface at 5 GHz (e.g. Ubiquiti NanoStation M2, TP-Link CPE 210, or the router mentioned here https://github.com/libremesh/lime-app/issues/324#issuecomment-924273982)

ilario, 2023-02-28:
This is important if we want to suggest FBW to non-LibreRouter users.
@selankon @spiccinini do you think that we can fix this before the 2020.2 release?