libremesh/lime-packages #706: Sysupgrade preserving lime-defaults config problems

spiccinini, 2020-05-05:
Currently using `sysupgrade` (or safe-upgrade) to upgrade a firmware while preserving configurations overwrites  unexpectedly the firmware `/etc/config/lime-defaults` with the current `lime-defaults` so it is not possible to provide a new default without using `-n` option of sysupgrade.

This happens because `/lib/upgrade/keep.d/base-files` provided by openwrt contains the following line `/etc/config/`. So all all /etc/config files are backup and preserved (including lime-defaults). 

There are multiple ways to fix. Some ideas:
* Moving /etc/config/lime-defaults to a `/usr/share/lime/config` directory (to more clearly imply that this file should not be edited). 
* Edit the keep.d/base-file removing the /etc/config line (explicitly declaring file per file which must be preserved)
* Do not use the backup facility of sysupgrade and provide another way.

ilario, 2020-05-05:
First proposed solution sounds good.
The new location should be pointed out in the comments in lime-node, lime-community, lime-example, and on the website.

dangowrt, 2020-06-02:
Yes, first proposal is the way to go imho. You can easily tell UCI to use an alternative configuration directory (we are already doing that in places, ie. for `/var/state/`)

spiccinini, 2020-06-03:
> Yes, first proposal is the way to go imho. You can easily tell UCI to use an alternative configuration directory (we are already doing that in places, ie. for `/var/state/`) and have lime-defaults loaded from an alternative location.
great (yes, in the unittests we are also faking the config directory creating an empty environment for each test)

> Also note that sysupgrade only preserves files in /etc/config if they have actually been modified.
Ah, didn't know that. Anyway this is a problem because not only we are modifying them from lime-config but also some are created durin first boot. This leads to many problems while upgrading to different versions of openwrt.

With @germanferrero we worked in a proposal to aproach configs in libremesh that will allow easier maintenance and it is related to this issue https://github.com/libremesh/lime-packages/issues/719

spiccinini, 2020-09-16:
Fixed in #738

closed 2020-09-16 by spiccinini