libremesh/lime-packages #709: Dnsmasq package recursive error in 19.07

spiccinini, 2020-05-13:
This is the error:
```
# make defconfig 
....
tmp/.config-package.in:73022:error: recursive dependency detected!                                                   
For a resolution refer to Documentation/kbuild/kconfig-language.txt                                                  
subsection "Kconfig recursive dependency limitations"                                                                                                 
tmp/.config-package.in:73022:   symbol PACKAGE_lime-proto-anygw depends on PACKAGE_dnsmasq              
For a resolution refer to Documentation/kbuild/kconfig-language.txt                                                  
subsection "Kconfig recursive dependency limitations"                                                                                                 
tmp/.config-package.in:173:     symbol PACKAGE_dnsmasq is selected by PACKAGE_dnsmasq-dhcpv6                         
For a resolution refer to Documentation/kbuild/kconfig-language.txt                                                  
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:187:     symbol PACKAGE_dnsmasq-dhcpv6 is selected by PACKAGE_lime-proto-anygw
```

`lime-proto-anygw` package has the following dependecies `DEPENDS:=+dnsmasq-dhcpv6 @(!PACKAGE_dnsmasq) ...`

Using `!PACKAGE` does not seem to work anymore as expected or something like that.  Also I don't see the usage of `!PACKAGE` in openwrt repositories, only in libremesh.

Does anyone have a clue about this?

ilario, 2020-05-14:
I remember @g10h4ck saying that we need dnsmasq-dhcpv6 and that it is incompatible with dnsmasq package... It would be great if @g10h4ck explains this.
In OpenWrt's menuconfig it is allowed to select both dnsmasq and dnsmasq-dhcpv6, logic would say that if we select both we'll have also the -dhcpv6 features, no need to deselect dnsmasq. But I didn't test if this actually happens.

spiccinini, 2020-05-14:
Selecting both at menuconfig si possible but at it will throw an error at build time (because no more that 1 package is allowed to have the same file). I don't know why the config does allow selecting both and if it is or not possible to explicitly mark as mutually exclusive.
So the idea was that selecting explicitly the anygw package would deselect the dnsmasq thing. In 18.06 what was happening is that if you selected dnsmasq then anygw would be deselected and "invisible" at menuconfig.

Maybe it is better to just drop this negated disallowance and just leave anygw depend on dnsmasq-dhcpv6.

ilario, 2020-05-14:
> Selecting both dnsmasq and dnsmasq-dhcpv6 at menuconfig si possible but it will throw an error at build time

LOL I didn't expect that!

> So the idea was that selecting explicitly the anygw package would deselect the dnsmasq thing.

That would be amazing but it never happened.

> In 18.06 what was happening is that if you selected dnsmasq then anygw would be deselected and "invisible" at menuconfig.

And yes, this is what always happened.
 
> Maybe it is better to just drop this negated disallowance and just leave anygw depend on dnsmasq-dhcpv6.

Mmh this sounds not much better than the current situation.
Would the clients not have an IPv6 if we remove this from anygw or what would be the drawback?

ilario, 2020-05-15:
Related issue: #331 and vaguely also #673

spiccinini, 2020-05-27:
> Mmh this sounds not much better than the current situation.
> Would the clients not have an IPv6 if we remove this from anygw or what would be the drawback?

My proposal is to do this:
```diff
--- a/packages/lime-proto-anygw/Makefile
+++ b/packages/lime-proto-anygw/Makefile
@@ -21,7 +21,7 @@ define Package/$(PKG_NAME)
     CATEGORY:=LiMe
     MAINTAINER:=Gioacchino Mazzurco <gio@eigenlab.org>
     URL:=http://libremesh.org
-    DEPENDS:=+dnsmasq-dhcpv6 @(!PACKAGE_dnsmasq) +ebtables +libuci-lua \
+    DEPENDS:=+dnsmasq-dhcpv6 +ebtables +libuci-lua \
         +lime-system +lua +kmod-ebtables +kmod-macvlan \
         +shared-state +shared-state-dnsmasq_leases +kmod-ebtables-ipv6
   PKGARCH:=all
```
What do you think?

ilario, 2020-06-05:
Yes, I agree, let's remove all kind of weird dependencies.
See also #716 #673 #331 #512 for issues generated by too many dependencies :)

G10h4ck, 2020-06-05:
> My proposal is to do this:
>
> ```diff
> --- a/packages/lime-proto-anygw/Makefile
> +++ b/packages/lime-proto-anygw/Makefile
> @@ -21,7 +21,7 @@ define Package/$(PKG_NAME)
>      CATEGORY:=LiMe [...]

I agree with that, at least now we get a build error if the wrong package is selected, before it was even more confusing because dnsmasq without IPv6 support would silently end up in the image

PR #755 (2020-09-01): lime-proto-anygw remove negative dependency
This change was originally proposed by @spiccinini and discussed on #709. 
Fixes #709
closed 2020-09-01 by G10h4ck