libremesh/lime-packages #720: shared-state pulls in 200kB of dependencies

dangowrt, 2020-06-04:
... and that even though uhttpd is already installed. Just adding shared-state eats 200kB.
```
Installing shared-state (2020-04-28-1588095602) to root...
Installing rpcd (2020-05-26-67c8a3fd-1) to root...
Installing libubus-lua (2019-12-27-041c9d1c-1) to root...
Installing luci-lib-jsonc (git-20.154.47914-8bf301a-1) to root...
Installing liblucihttp0 (2019-07-05-a34a17d5-1) to root...
Installing liblucihttp-lua (2019-07-05-a34a17d5-1) to root...
Installing rpcd-mod-file (2020-05-26-67c8a3fd-1) to root...
Installing rpcd-mod-luci (20191114) to root...
Installing cgi-io (19) to root...
Installing luci-base (git-20.154.47914-8bf301a-1) to root...
Installing luci-lib-httpprotoutils (git-20.154.47914-8bf301a-1) to root...
```
Why is all this needed just for a bit of shared-state? Do you assume that web-ui is always installed?
Previously, I was able to use alfed-based leased sharing even on very small boxes with no problem, which unlocked anygw. Now not any more.
Any chance to use the old alfred-based lease propagation (it always worked well for me with recent enough alfred)?
Or if even alfred is overkill, I volunteer to write a small TCP service in C which does the job in less than 10kB (without any Lua or HTTP involved).

ilario, 2020-06-04:
Currently shared-state package depends on:

https://github.com/libremesh/lime-packages/blob/94b67f51fce8c92d943e93477e03efb7b0cfd45e/packages/shared-state/Makefile#L22-L23

while [luci-lib-nixio](https://openwrt.org/packages/pkgdata/luci-lib-nixio) and [luci-lib-jsonc](https://openwrt.org/packages/pkgdata/luci-lib-jsonc) depend on small things, [luci-lib-httpclient](https://openwrt.org/packages/pkgdata/luci-lib-httpclient) depends on luci-base which pulls in loads of stuff.

We could maybe use Busybox's netcat instead of luci.httpclient for doing post?

https://github.com/libremesh/lime-packages/blob/1000d76a79472044a063b59003c730fda9c72209/packages/shared-state/files/usr/lib/lua/shared-state.lua#L182-L189

dangowrt, 2020-06-04:
yeah, luci-lib-jsonc is not a problem, neither is nixio.
Maybe use uclient-fetch for the POST request?
It's anyway already installed and needed for many OpenWrt features (we dropped busybox-wget a while ago in favour of uclient-fetch, in that way HTTPS is possible using libustream-*)
```
Usage: uclient-fetch [options] <URL>
Options:
	-4				Use IPv4 only
	-6				Use IPv6 only
	-q				Turn off status messages
	-O <file>			Redirect output to file (use "-" for stdout)
	-P <dir>			Set directory for output files
	--user=<user>			HTTP authentication username
	--password=<password>		HTTP authentication password
	--user-agent|-U <str>		Set HTTP user agent
	--post-data=STRING		use the POST method; send STRING as the data
	--spider|-s			Spider mode - only check file existence
	--timeout=N|-T N		Set connect/request timeout to N seconds
	--proxy=on|off|-Y on|off	Enable/disable env var configured proxy

HTTPS options:
	--ca-certificate=<cert>		Load CA certificates from file <cert>
	--no-check-certificate		don't validate the server's certificate

```

ilario, 2020-06-04:
I didn't know uclient-fetch, is sounds perfect!
@G10h4ck what do you think?

G10h4ck, 2020-06-04:
> I didn't know uclient-fetch, is sounds perfect!
> @G10h4ck what do you think?

Neither did I!
Seems promising, right now I don't remember exactly which features we needed about `luci-http-client`, probably we used to set some short timeout but since `luci-http-client` is buggy adn depends on big stuff I am quite happy to substitute it!

dangowrt, 2020-06-04:
uclient-fetch (aka. wget in OpenWrt) is a CLI frontend to a shared library libuclient which uses the usual OpenWrt libubox/blobmsg/... tooling to integrate nicely with all the other things in the OpenWrt universe (ubus, blobmsg-json, ...). And drop-in TLS support with libustream which is used in the same way by uhttpd to provide TLS support and links against either openssl, mbedtls or wolfssl.
If you need any help making good use of any of that, feel more than welcome to ask me.

PR #722 (2020-07-08): shared-state: use uclient-fetch instead of luci-lib-httpclient
luci-lib-httpclient pulls in 200kB of dependencies.
Replace it with a wrapper around uclient-fetch.

Fixes #720

(I'm not much of a Lua hacker...)
closed 2020-07-08 by G10h4ck