libremesh/lime-packages #741: lime-hwd-openwrt-wan conflicts with lime net in the wan interface

spiccinini, 2020-08-14:
I tried to configure a static config in the interface `eth0.1`  as follows:
```
        config net 'ethstatic'
            option linux_name 'eth0.1' # the WAN ifc of the librerouer
            list protocols 'static'
            option static_ipv4 '10.62.99.99/16'
            option static_gateway_ipv4 '10.62.0.2'
```
But this does not work (it does not generate a static config). I tracked the problem down to `lime-hwd-openwrt-wan`. This configures the same interface and marks the 'wan' protocol.

This is beacause the interface is "preconfigured" twice here:
```
function network.configure()
	local specificIfaces = {}

	config.foreach("net", function(iface)
		if iface["linux_name"] then
			specificIfaces[iface["linux_name"]] = iface
		end
	end)
```
and the last (openwrt-wan) wins.

I don't know what to do to fix this.

spiccinini, 2020-08-14:
@G10h4ck @ilario I assigned you just because I think that maybe you know what to do. I can do the work if you point me what would be a good way to solve this, or even if this is something to solve or not.

G10h4ck, 2020-08-25:
AFAIK lime-hwd-openwrt-wan should run before that as it is an hardware detection module

ilario, 2020-09-22:
Hey @spiccinini can you confirm that setting the autogenerated to false (as I documented a few days ago here: https://github.com/libremesh/lime-packages/commit/9cc600a135608474911c5207c2d1fed082e11954#diff-cfd09ba8b27300468cac8308fc444415R151-R152 ) solves the issue?

The easy question now is: is this documented well enough? Clearly not. And where should we document it?