libremesh/lime-packages #747: Removing lime-community and lime-node files from lime-system package

ilario, 2020-08-24:
This is more a question than a proposal:
why the `/etc/config/lime-community` and `/etc/config/lime-node` are included in `lime-system` package?

Isn't this likely going to conflict with their versions included in the [network-profiles](https://github.com/libremesh/network-profiles/find/master) if the network-profiles are included at build time (either from source via buildroot or as packages via SDK)?

Wouldn't be better to not forcefully require them (so they would not exist unless a user creates them) or to create them at the first boot (if they are really needed)?

spiccinini, 2020-08-24:
> This is more a question than a proposal:
> why the `/etc/config/lime-community` and `/etc/config/lime-node` are included in `lime-system` package?

Well I see some reasons for thouse files to be in a package (not necesarly lime-system):
* UCI needs the files to be present to be able to "use or modify them" (with uci set, etc).
* In general is good to have all the files in the system defined in a package. This simplifies discoverability, thinking about the system, ease maintenance.

> Isn't this likely going to conflict with their versions included in the [network-profiles](https://github.com/libremesh/network-profiles/find/master) if the network-profiles are included at build time (either from source via buildroot or as packages via SDK)?

If the files are included from source using the `files/` directory it won't be a conflict (as everything in this directory overwrites what it is in the packages). But yes it will conflict if using the network-profiles through packages via SDK. I see that there may be interest in installing and upgrading the network-profiles through opkg, I don't use this way.

> Wouldn't be better to not forcefully require them (so they would not exist unless a user creates them) or to create them at the first boot (if they are really needed)?

In my opinion If they don't exist in the system it would be a lot more difficult to modify them in the target.
Creating them at the first boot is not ideal, it adds more complexity.

It would be good to be able to use the opkg Provides and Replaces for this.
For example we may create a `lime-config` package that `PROVIDES:=libremesh-config` and then it would be able to have 
packages for each network like `network-profile-foo` that `PROVIDES:=libremesh-config` and `REPLACES:=lime-config`.
I did a test of this and it works but openwrt ipkg glue does not support `REPLACES` :(  I don't understand why as openwrt's opkg do supports Replaces, it is just a limitation of the buildroot glue that creates the metadata of the package.

WIth this patch it works:
```
--- a/include/package-ipkg.mk
+++ b/include/package-ipkg.mk
@@ -159,6 +159,7 @@ Version: $(VERSION)
 $$(call addfield,Depends,$$(Package/$(1)/DEPENDS)
 )$$(call addfield,Conflicts,$$(call mergelist,$(CONFLICTS))
 )$$(call addfield,Provides,$$(call mergelist,$(PROVIDES))
+)$$(call addfield,Replaces,$$(call mergelist,$(REPLACES))
 )$$(call addfield,Alternatives,$$(call mergelist,$(ALTERNATIVES))
 )$$(call addfield,Source,$(SOURCE)
 )$$(call addfield,License,$(LICENSE)
```
Maybe if there is interest we can propose the patch to OpenWrt.

ilario, 2020-08-24:
I didn't know the PROVIDE and REPLACE, they sound perfect :D 

Maybe the package should not be named `lime-config` but `lime-config-dummy` or something making explicit that it does not have any actual content.

Currently, I expect a conflict:

* when using a SDK (e.g. Chef);
* when using OPKG (as I did [here](https://blog.freifunk.net/2019/06/24/load-correlated-distributed-bandwidth-analysis-for-libremesh-networks-2-setting-up-the-libremesh-test-network/) and as was possible using the network-profiles packages repository which was available [here](https://snapshots.libremesh.org/packages) and which will likely become available again once [this](https://github.com/libremesh/network-profiles/actions) actually works);
* when using network-profiles sources inside feeds.conf (which in my opinion is way cleaner than using the `files/` directory) see the discussion [here](https://github.com/libremesh/network-profiles-builder/issues/2).

On the contrary it does not cause a conflict when using the `files/` directory (as @spiccinini correctly pointed out).

ilario, 2020-09-21:
I would consider this as a high priority issue, but this also depends on which Chef-like or SDK we will support and how these implement the inclusion of the network-profiles.

spiccinini, 2020-10-11:
I opened #774 as one way to fix this.

Another way may be to use a `preinst` script in the network-profiles Makefiles to remove the config files provided by lime-system.

ilario, 2020-10-12:
Thanks @spiccinini amazing work!
#774 would allow to compile including a network-profile using the buildroot and using an SDK (none is currently supported).
As a much lower priority issue we could consider to support the installation through OPKG, which will require the preinst script to be included in [profile.mk](https://github.com/libremesh/network-profiles/blob/master/profile.mk), but this is maybe not even needed/not even an issue.

ilario, 2020-10-15:
To solve the installation through OPKG is as easy as selecting "Overwrite files from other package(s)" on LuCI or using `--force-overwrite` opkg option in command line.

ilario, 2020-10-18:
Closing thanks to #774

closed 2020-10-18 by ilario