libremesh/lime-packages #750: make warning bird4 and bird6 does not exist

rallep71, 2020-08-28:
Hello, 
I have tried to build Libremesh from the documentation on the website of Libremesh.

as a base i used openwrt 19.07.3.

i got the following error

make -j$(nproc)
WARNING: Makefile 'package/feeds/libremesh/lime-proto-bgp/Makefile' has a dependency on 'bird4', which does not exist
WARNING: Makefile 'package/feeds/libremesh/lime-proto-bgp/Makefile' has a dependency on 'bird6', which does not exist

what can I do?

Translated with www.DeepL.com/Translator (free version)

ilario, 2020-08-28:
Thanks for the report!
This is a known problem which is being tracked in this issue: #645 
Anyway it is just a warning (not an error) and has no practical consequences for LibreMesh users, so you can safely ignore that nessage.
Closing as duplicate of #645

closed 2020-08-28 by ilario

rallep71, 2020-08-28:
Thx ! :dancers: