libremesh/lime-packages #752: No valid theme found

rallep71, 2020-08-28:
Hello,
Sorry to bother you again,

i have successfully created and flashed the image, now i get the following message
/usr/lib/lua/luci/dispatcher.lua:724: No valid theme found
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:724: in function 'init_template_engine'
	/usr/lib/lua/luci/dispatcher.lua:848: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

openwrt 19.07.3

https://libremesh.org/development.html

ilario, 2020-08-28:
Hi!
Wow, this issue is back again, it happened plenty of times and I thought we finally got rid of it...
See #609 fixed in #651, then #689 fixed in cecfba70b0cc6e1759a58d8e48a53496b6e517ae and again in f4f8d004ff8564452ab705b02687eff9cc063334 and fixed in c88496cfaea09acc6e8b93d91c3c4e529d5d2f29.

This time I don't know why this is happening (and I did not test on my devices if it happens also to me), the [index file](https://github.com/libremesh/lime-packages/blob/master/packages/lime-app/files/www/lime_app_index.html) looks ok and also [the one setting the uci option](https://github.com/libremesh/lime-packages/blob/master/packages/lime-app/files/etc/uci-defaults/96-lime-app-index_page).

Can you confirm that you selected the lime-app package during the compilation?
Can you log in your router via ssh and confirm that the some files exist, maybe using the following commands?
```
cat /www/lime_app_index.html
cat /etc/uci-defaults/96-lime-app-index_page (should it be there or does it autodestroy after first boot?)
ls /www/app/
uci get uhttpd.main.index_page
```
Can you try to manually specify `/app` in the URL after the direction of your router in your browser? Something like http://thisnode.info/app
Thanks!

rallep71, 2020-08-28:
Danke für die schnelle Antwort,

ja, ich habe die lime-app ausgewählt <*>
und ja, ich habe ssh root zugriff
cat /www/lime_app_index.html

<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.1//EN" "http://www.w3.org/TR/xhtml11/DTD/xhtml11.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
<head>
<meta http-equiv="Cache-Control" content="no-cache" />
<meta http-equiv="refresh" content="0; URL=/app" />
</head>
<body style="background-color: white">
<a style="color: black; font-family: arial, helvetica, sans-serif;" href="/app">LiMe-App</a>
</body>
</html>

next

root@LiMe-4563ac:/# cat /etc/uci-defaults/96-lime-app-index_page

cat: can't open '/etc/uci-defaults/96-lime-app-index_page': No such file or directory

next

root@LiMe-4563ac:/# ls -l /www/app/

drwxr-xr-x    3 root     root            44 May 16 18:32 assets
-rw-r--r--    1 root     root        282630 May 16 18:32 bundle.42d2e.js
-rw-r--r--    1 root     root         11225 May 16 18:32 bundle.98dff.css
-rw-r--r--    1 root     root          1069 May 16 18:32 index.html
-rw-r--r--    1 root     root           303 May 16 18:32 manifest.json
-rw-r--r--    1 root     root          5155 May 16 18:32 polyfills.d3f7d.js
-rw-r--r--    1 root     root            11 May 16 18:32 preact_prerender_data.json
-rw-r--r--    1 root     root           928 May 16 18:32 precache-manifest.8d44203f78917a379daccc5f7a807bd5.js
-rw-r--r--    1 root     root           101 May 16 18:32 push-manifest.json
-rw-r--r--    1 root     root           709 May 16 18:32 sw-debug.js
-rw-r--r--    1 root     root          1690 May 16 18:32 sw.js

next

root@LiMe-4563ac:/# uci get uhttpd.main.index_page
lime_app_index.html

Can you try to manually specify /app

yes i can
[image]

ilario, 2020-08-29:
All the commands' output seems good.

> Can you try to manually specify /app
> 
> yes i can
> [image]
> 
> Important to me is not so much the redirect on lime app "/app", I can also enter that by hand.

Ok, so we will keep this part of the discussion on #609
Anyway, once the redirect towards .../app works, what you will have to type by hand will be .../cgi-bin/luci in order to access the traditional interface.

> it is important to me that I can also access the luci web interface to change passwords etc...
> 
> i can also do that in the cli, but that is a little more complicated.

Ok, so you'll have to install also the traditional web interface.

Sincerely I cannot remember which packages you need to install.
@digitigrafo you use both interfaces, right?
I suspect that the LibreMesh customization to LuCI is outdated, so the safest option is to simply install the LuCI collection (the package is simply named `luci` and you can find it in menuconfig under LuCI/1. Collections/luci).

rallep71, 2020-08-30:
Thanks for the answers.

I have now built different binerys again, now everything works.

No idea what didn't work when compiling.

Thanks anyway.

ilario, 2020-08-30:
:)

closed 2020-08-30 by ilario