libremesh/lime-packages #758: changes faster into the new node

rallep71, 2020-09-08:
Hello, how can I realize that in the 2,4ghz node connected my smartphone changes faster into the new node if the signal from the connected node gets worse? this sometimes takes up to 15 - 20 seconds, in this time my smartphone changes into lte. 

In 5ghz node connection this is no problem, it works wonderfully.

All three nodes are also connected by cable.

could you give me an example of how I could improve this?

ilario, 2020-09-08:
Super interesting question!
I have no idea how to do that, but we could investigate again #253

rallep71, 2020-09-08:
Thank you for the quick response
I took a look at this, 

https://gitlab.com/slthomason/freemesh/-/blob/master/package/freemesh/cfg-router/files/etc/config/wireless

which additional options would be useful to include in radio0 libremesh?

option noscan '1'
option short_preamble '1'
option disabled '0'

and in config wifi-iface 'lm_wlan0_ap_radio0'

option r1_key_holder ''
option ft_over_ds '1'
option ft_psk_generate_local '1'
option ieee80211r '1'

?
what would be useful and what do the options do?
ieee80211r is roming, as far as is clear.

would i destroy my meshnetwork with it? does it affect the whole system libremesh?

ilario, 2020-09-08:
I don't know, maybe @nordurljosahvida @openspace42 tested 802.11r?

ilario, 2023-02-28:
No news since 2020, so maybe we have no way or no idea how to fix that?
Closing, but feel free to re-open if you have proposals.

closed 2023-02-28 by ilario