libremesh/lime-packages #763: Lime-App, Un unexpected error occurred.

rallep71, 2020-09-20:
Lime-App, Un unexpected error occurred. 

Hello,

once again I have a problem with the display in the browser of the Lime-App. I don't need the First-Boot-Wizzard, that's why it is not compiled. I only need to see in which node I am located in the Lime-App. See photo. 

systems:
Openwrt 19.07.4, following routers are in use, TP-Link Archer C50 v3, TP-Link Archer C50 v4, TP-Link TL-WDR4300 v1.

Config File:
https://pastebin.com/sdpGx1Ca
feeds.conf:
src-git packages https://git.openwrt.org/feed/packages.git^eae1bb397f064befe003b01a36ae318321e81d35
src-git luci https://git.openwrt.org/project/luci.git^0d0ab01a64f25d15325d3c98a7df252085382f1d
src-git routing https://git.openwrt.org/feed/routing.git^b77498bd56d5e45ab4577a1f4ad6ffc55b4a86b7
src-git telephony https://git.openwrt.org/feed/telephony.git^91b34e7eefa1d37a18390d0c6881978cd76804a1
src-git libremesh https://github.com/libremesh/lime-packages.git
src-git libremap https://github.com/libremesh/libremap-agent.git
src-git networkprofiles https://github.com/libremesh/network-profiles.git

[image]

have I forgotten to compile anything, or what could I do?

THX

spiccinini, 2020-09-20:
Hi @rallep71 indeed the lime-app error is related to firstbootwizard not being installed. Thanks for reporting it :)

In order to use encrypted mesh, the wpad-mini package have to be replaced with wpad-mesh-wolfssl package.

rallep71, 2020-09-21:
Thanks for the quick answer!
I had mentioned it before in another posting, I am very impressed with libremesh! I have been looking for solutions to build a mesh for over a year. Sure, there are ready-made solutions from different vendors including hardware, but I don't have control over them, and usually after two or three years there are no more updates. 

I replaced wpad-mini with wpad-mesh-openssl, what is the difference to wpad-mesh-wolfssl? would wpad-mesh.openssl also be possible?

again to my understanding, is it possible to realize this without fbw? would both ways to view the node info be possible? or even a minimal version which only shows in which node you are?

the documentation on the website is kept very minimal, it is enough to build a mesh for the beginning. I had to make many attempts, i.e. compiling new images with more software, until I am now on this stand to know what is really needed, also with regard to security. maybe there would be some need for action.

AndyMcSchopf, 2020-09-22:
Hi all,

I have a similar Problem - same Error-Message, and encrypted Mesh.
The differnce is, that I have fbw compiled into my images.

Maybe there is an other problem here?

P.S. @rallep71 afaik wolf-ssl is a much smaller implementation - my images are about 1M smaller compared to openssl...

rallep71, 2020-09-22:
Hello everybody,

i have done my work again and compiled a new image, this time with fbw.

So far so good. Now I don't get any more errors, Lime-App is displayed as it should be. except for a small error but more about that in the next posting.

I was really amazed for the first time! Not in a positive sense.

I gave the router a root password, I encrypted the AP's, also the mesh.

I log on to the AP, go to the node, fwb, create a new network. So I can create a new network, no matter if the network is encrypted or not.

I create a new network with fbw, assign a password, assign the network name, that's it..I am not asked if I am allowed to do this at all, it's easy. 

the new network is created and at the same time with this new password the root password is changed, unbelievable. And, the whole original configuration is gone, too. Guys, this makes no sense with the fbw, honestly not.

ilario, 2020-09-22:
The reported issue about FirstBootWizard looks like #640 (and in my opinion it can be more serious when considering also #280), which was closed based on the fact that FBW has an option for disabling it when configuring the system manually.
@rallep71 if you agree that #640 describes your case, could you continue the thread there?

rallep71, 2020-09-22:
Thanks for the answer, 

yes of course I agree that we continue there #640, that's fine. oder auf deutsch, des passt scho :)

thx

AndyMcSchopf, 2020-09-23:
> 
> 
> Hi all,
> 
> I have a similar Problem - same Error-Message, and encrypted Mesh.
> The difference is, that I have fbw compiled into my images.
> 
> Maybe there is an other problem here?
> 
> P.S. @rallep71 afaik wolf-ssl is a much smaller implementation - my images are about 1M smaller compared to openssl...

I tried the whole thing again - made new images - but still have the problem. 
I used 19074 with feeds:
src-git packages https://git.openwrt.org/feed/packages.git^eae1bb397
src-git luci https://git.openwrt.org/project/luci.git^0d0ab01a6
src-git routing https://git.openwrt.org/feed/routing.git^b77498b
src-git telephony https://git.openwrt.org/feed/telephony.git^91b34e7
src-git libremesh https://github.com/libremesh/lime-packages.git^80c79378
src-git libremap https://github.com/libremesh/libremap-agent.git^e2b94dc

Not working on meraki mx60, mr18, ubiquity ac-mesh and wavlink 577a2... 
Still getting the error message above.

rallep71, 2020-09-23:
hi Andy ,

make clean, 
scripts/feeds update -a
scripts/feeds install -a
make manuconfig
select > LiMe select first-boot-wizard , safe exit
select > Ubus > select 3. Aplications, select lime-ubus-fbw, safe exit
make download
make...

AndyMcSchopf, 2020-09-23:
Hi rallep71,

i will check with your instructions tomorrow...

Danke Dir! ;-)

AndyMcSchopf, 2020-09-24:
Ok so the point is you need both - lime first boot wizard and the ubus-lime-fbw.
Then the Lime App is woking - but every time you start it it wants to create a new network...

This whole behaviour changed in the last 2 weeks or so - I had older builds with an (more or less) working lime app without that ubus-lime-fbw... Hm

AndyMcSchopf, 2020-09-24:
Update:
with the info mentioned here: [https://github.com/libremesh/lime-packages/issues/640#issuecomment-558081082](url) I added option firstbootwizard_configured 'true' to lime-node.
Now the fbw is away and the configs need the knowledge of the root password.

rallep71, 2020-09-24:
hm, i have done the same now, but no difference, fbw is still there and the configuration can still be changed.

rallep71, 2020-09-24:
Hello, 
I stand corrected, but what procedure...
with fbw first set something, rootpw, node..etc, then a short reboot, and only then change everything in luci e.g. mesh and ap encryption etc.. 

I can live with that for now :D

thx andy ;)

AndyMcSchopf, 2020-09-24:
You're welcome ;-)
I solved the whole config thing with an lime-community and lime-node file compiled in the firmware (for each with own config like lime-node-mr18, lime-node-mx60, etc.) and then only rename on the router -> lime-config, lime-apply and voila...

AndyMcSchopf, 2020-09-24:
BTW for the `firstbootwizard_configured 'true' ` to work you have to reboot. For the most other things the `lime-config `and `lime-apply `commands are sufficiant...

rallep71, 2020-09-24:
you'll have to explain this to me again, because that's where I'm standing on the hose right now..

jetzt scheint batman nicht zu laufen, ich bekomme noch die krise...
[image]

AndyMcSchopf, 2020-09-24:
Ok short in german:

Ich habe die gesamte Konfiguration nicht über luci sondern über die Config-Files von libremesh gemacht.
Also auf der Konsole (ssh auf den node1) unter dem Verzeichnis `/etc/config` die Dateien lime-community und lime-node angepasst.
`lime-community `ist das config-File für alle nodes - und `lime-node ` dann noch Node-spezifisch.
Wird hier genauer erklärt:
-> [https://libremesh.org/docs/en_config.html](https://libremesh.org/docs/en_config.html)

Wenn die angelegt sind wird über die Befehle `lime-config `die Konfiguration komplett automatisch gebaut und mit `lime-apply` direkt angewandt. Wie gesagt bei dem firstbootwizard kommt noch der reboot dazu.

Bei Bedarf kann ich dir auch meine zwei configs mal auflisten...

rallep71, 2020-09-24:
Danke Dir, 
meine Fragen wären, die lime-community und die lime-node bestückst Du erst nach dem ersten Bootvorgang gemeinsam mit  fbw auf true? dann lime-config + lime-apply und dann reboot?
hast Du die selbe feeds.conf wie ich? oder setzt du manuell wpad-mesh-wolfssl in menuconfig? um mesh zu verschlüsseln?

Sehr gerne würde ich mir deine Dateien ansehen, Danke Dir sehr!!

Thank you, 
my questions would be, the lime-community and the lime-node you only populate after the first boot process together with fbw to true? then lime-config + lime-apply and then reboot?
do you have the same feeds.conf as me? or do you manually set wpad-mesh-wolfssl in menuconfig? to encrypt mesh?

I would love to have a look at your files, thank you very much!

AndyMcSchopf, 2020-09-24:
> 
> 
> Danke Dir,
> meine Fragen wären, die lime-community und die lime-node bestückst Du erst nach dem ersten Bootvorgang gemeinsam mit fbw auf true? dann lime-config + lime-apply und dann reboot?

Jepp, genau so (bzw. kopieren der im image enthaltenen lime-community-myNet Datei auf lime-community, gleiches bei der node) ..

> hast Du die selbe feeds.conf wie ich? oder setzt du manuell wpad-mesh-wolfssl in menuconfig? um mesh zu verschlüsseln?
> 

feeds.conf analog der Dokumentation auf [libremesh.org](https://libremesh.org/development.html).
wpad-mesh-wolfssl setze ich im menuconfig.

> Sehr gerne würde ich mir deine Dateien ansehen, Danke Dir sehr!!
> 
> Thank you,
> my questions would be, the lime-community and the lime-node you only populate after the first boot process together with fbw to true? then lime-config + lime-apply and then reboot?

yes
> do you have the same feeds.conf as me? or do you manually set wpad-mesh-wolfssl in menuconfig? to encrypt mesh?

feeds.conf as described on [libremesh.org](https://libremesh.org/development.html)
wpad-mesh-wolfssl set in menuconfig
> 
> I would love to have a look at your files, thank you very much!

example of my lime-community file:
```
### System options

config lime system
        option hostname 'myNet-%M4%M5%M6'
        option domain   'myNet.local'

### Network general option

config lime network

### WiFi general options

config lime wifi
        option ap_ssid 'myNet'
        option ap_encryption 'sae-mixed'
        option ap_key '<key>'
        option apname_ssid 'myNet/%H'
        option apname_encryption 'sae-mixed'
        option apname_key '<key>'
        option adhoc_ssid 'myNet_adhoc'
        option ieee80211s_mesh_id 'myNet_mesh'
        option ieee80211s_encryption 'sae'       
        option ieee80211s_key '<key>'
```

Example for the lime-node on my MX60

```
config lime 'system'
        option hostname 'myNet-MX60'
	option firstbootwizard_configured 'true'

config lime 'network'

config lime 'wifi'
```

rallep71, 2020-09-24:
Danke Dir, soweit verstanden.
noch ein paar Fragen:

lime-community >> 
### Network general option

config lime network

lässt Du leer, wird das von der default übernommen?

2,4 Ghz und 5Ghz Netzwerk wären dann identisch, oder hast du nur ein 2Ghz Netzwerk laut Deiner Config?
Ich hatte im Nachgang immer unterschieden, zwischen 2 und 5, z.b. HTMesh2G  HTMesh2G/Node, HTMesh5G  HTMesh5G/Node, auch Mesh hatte ich aufgeteilt, HTMesh2G und HTMesh5G, 

Macht das überhaupt Sinn? 

Thank you, so far understood.
a few more questions:

lime-community >> 
### Network general option

config lime network

if you leave it blank, will it be taken over by the default?

2.4 Ghz and 5Ghz network would be identical, or do you have only a 2Ghz network according to your config?
I had always distinguished between 2 and 5, e.g. HTMesh2G HTMesh2G/Node, HTMesh5G HTMesh5G/Node, I had also divided mesh, HTMesh2G and HTMesh5G, 

Does that make any sense at all? 

Translated with www.DeepL.com/Translator (free version)

spiccinini, 2020-09-24:
@rallep71 Yes, blanks in the config are configured using the lower hierarchy config. lime-node > lime-community > lime-defaults

Here is a work in progress (in spanish and a bit of english) docs for the config system:  https://hackmd.io/CWm5MbwrToevK1-mLh4Y-Q?view I want to finalize it for the release but it may be useful for you as it is now.

Essids can be different or the same on 2.4 and 5ghz... depending on the use case. In our communities very few people have computers and phones with 5ghz so we use 5ghz only for the mesh and 2.4 ghz only for the wireless access to clients.

AndyMcSchopf, 2020-09-24:
@rallep71 it is as stated by spiccinini - normally I have identical 2 and 5 Ghz Nodes. 
On some nodes I change this by using the lime-node like this:

```
config lime 'network'

config lime 'wifi'

config wifi 'radio0'
        list modes 'ap'
        list modes 'ieee80211s'

config wifi 'radio1'
        list modes 'ap'
        list modes 'ieee80211s'

config wifi 'radio2'
        option channel '48'
        list modes 'ap'
        list modes 'apname'
```
radio0 is 2Ghz (mesh and ssid), radio1 5Ghz with 300M (mesh and ssid) 
and radio2 is dualband - which I configured to 5Ghz (150M) without mesh but the local node (apname)
just for example ;-)

rallep71, 2020-09-29:
Hello everyone, me again,
if i enter my configuration in lime-community and lime-node as described above, i don't get anything displayed in luci under batman, the image i posted above, why is that? I mean, it is good to see in luci what is going on without working with batctl in the terminal. If I put it back, it will be displayed again.

then a few more questions: should you run batman-adv in autogateway mode or not? the same option is also available under bmx6, what should you choose?

An ebtables[3] rule in the LAN/AP bridge prevents packets sent to the anycast address from spreading through the cloud. The mesh node to which the client is assigned receives the packet, but not the other nodes.
something like iptables, but for the network layer 2, 
what can I do with it?

in deutsch:)

Hallo zusammen, ich mal wieder,
wenn ich meine configuration wie weiter oben beschrieben in lime-community und lime-node eingebe bekomme ich in luci unter batman nichts mehr angezeigt, das selber bild wie ich weiter oben schon gepostet habe, warum ist das so? Ich meine, es ist in luci gut zu sehen was grad geht ohne im terminal mit batctl zu arbeiten. Stell ich wieder zurück, wird es wieder angezeigt.

Dann gleich noch ein paar Fragen: sollte man batman-adv im autogateway modus betreiben oder nicht? die selbe option gibt es ja auch unter bmx6, was wählt man?

Eine ebtables[3]-Regel in der LAN/AP-Bridge verhindert, dass Pakete, die an die Anycast-Adresse gesendet werden, sich durch die Cloud verbreiten. Der Mesh-Knoten, dem der Client zugeordnet ist, erhält das Paket, aber nicht die anderen Knoten.
etwas wie iptables, aber für die Netzwerkschicht 2, 
was kann ich damit anfange?

ilario, 2020-09-29:
> if i enter my configuration in lime-community and lime-node as described above, i don't get anything displayed in luci under batman, the image i posted above, why is that? I mean, it is good to see [...]

Mmh, it's loong time I don't use LuCI, so I cannot help here.
Maybe someone developing the [lime-app](https://github.com/libremesh/lime-app) can tell us if a similar feature is/will be present in lime-app? @germanferrero @gmarcos87 

> then a few more questions: should you run batman-adv in autogateway mode or not? the same option is also available under bmx6, what should you choose?

I don't understand what are you referring to, where is this option?
Do you refer to the batman-adv-auto-gw-mode and bmx6-auto-gw-mode packages on the LibreMesh repository?

> An ebtables[3] rule in the LAN/AP bridge prevents packets sent to the anycast address from spreading through the cloud. The mesh node to which the client is assigned receives the packet, but not the other nodes.
> something like iptables, but for the network layer 2,
> what can I do with it?

Do you want or do you not want such a rule?

Are you referring to this rule here?
https://github.com/libremesh/lime-packages/blob/92a98fb6f9f9796bcdf26513efa8f5290ea34b56/packages/lime-proto-anygw/files/usr/lib/lua/lime/proto/anygw.lua#L71

ilario, 2020-09-29:
I can confirm the error which originated this ticket:
I flashed a TP-Link WDR3600 with a recently compiled image [following the instructions on the website](https://libremesh.org/development.html#compiling_libremesh_from_source_code) based on OpenWrt 19.07.4 and got the error "Un unexpected error occurred, please contact the developer team".

So I tried to trace why it happens and seems that it happens if `safe-upgrade` package is not installed.

More of less the traceback is:
https://github.com/libremesh/lime-app/blob/bbb59f27eb11061474ef4ee2e49da43535789e5e/src/utils/app.context.js#L64

fails because here the value of the variable is -1 so that an error is thrown

https://github.com/libremesh/lime-packages/blob/0d539fe7fba6363355095170d6ac5803642f5871/packages/ubus-lime-utils/files/usr/libexec/rpcd/lime-utils#L190

and the value is -1 because a few code lines above it was set so by 

https://github.com/libremesh/lime-packages/blob/0d539fe7fba6363355095170d6ac5803642f5871/packages/ubus-lime-utils/files/usr/libexec/rpcd/lime-utils#L185

which calls 

https://github.com/libremesh/lime-packages/blob/0d539fe7fba6363355095170d6ac5803642f5871/packages/ubus-lime-utils/files/usr/libexec/rpcd/lime-utils#L167-L169

which cannot find the `safe-upgrade` executable as it is not installed.

germanferrero, 2020-09-29:
> Maybe someone developing the [lime-app](https://github.com/libremesh/lime-app) can tell us if a similar feature is/will be present in lime-app? @germanferrero @gmarcos87
Hi @ilario there is something like this in the roadmap. We will be adding (not soon though), a comprehensive map of the network, integrating some of the features that were available at [libre-map](https://github.com/libremap) into the [lime-app](https://github.com/libremesh/lime-app). Our goal is to provide a graphical tool for network diagnosis for non-geeks ppl.

germanferrero, 2020-09-29:
@ilario thanks for reviewing and debugging the code. Although I must admit that this issue arose because I didn't took into account the possibility of not including fbw packages in the firmware. It's now being tracked at https://github.com/libremesh/lime-app/issues/275 as @spiccinini stated.

germanferrero, 2020-09-29:
Also, in answer to @rallep71, thanks for discovering this _big_ bug. We are actually allowing for a privilege escalation with the current implementation of the firstbootwizard process. Which is obviously, an unintentional mistake.
I have done a step forward in solving it at: https://github.com/germanferrero/lime-app/commit/db6bfe4eecbdb839cb58368ca0cd25afcfee2893
Which implements asking for root password before entering fbw if the password has been set.
It should be accompanied with the changes at the acl for fbw at ubus-lime-fbw.
I hope we address it very soon.
Apologizes

ilario, 2020-10-10:
My traceback of the error was wrong, sorry for the mistake.

ilario, 2020-10-30:
Should have been fixed in https://github.com/libremesh/lime-app/pull/279 but this will be present in lime-app 0.2.10 whilst the version pulled by lime-packages is still the 0.2.9:
https://github.com/libremesh/lime-packages/blob/3a20c6ad5ddf35990d7ec22d03628c1257537212/packages/lime-app/Makefile#L9

@spiccinini @germanferrero do you think we can have the Makefile pointing to the 0.2.10?

germanferrero, 2020-10-30:
https://github.com/libremesh/lime-packages/pull/793#event-3941864585 have been merged, so I close this issue. Feel free to re-open if necessary

ilario, 2020-11-01:
I just tested and still got the same error. @germanferrero @spiccinini does it work for you?
I don't know anything of JS but looks like this error handling does not work:
https://github.com/libremesh/lime-app/blob/v0.2.10/src/utils/app.context.js#L84-L85

spiccinini, 2020-11-01:
> I just tested and still got the same error. @germanferrero @spiccinini does it work for you?
> I don't know anything of JS but looks like this error handling does not work:
> https://github.com/libremesh/lime-app/blob/v0.2.10/src/utils/app.context.js#L84-L85

I have tested and it is working for me. Maybe it is another part of the initialization that is failing to you?
I have this response `{"jsonrpc":"2.0","id":4,"error":{"code":-32000,"message":"Object not found"}}` to the  the `lime-fbw status` call and the page shows no error. 
Maybe the problem is other call?

ilario, 2020-11-01:
> I have tested and it is working for me.

So, lime-app is opening for you even without lime-fbw and ubus-lime-fbw?

> Maybe it is another part of the initialization that is failing to you?
> Maybe the problem is other call?

Nope, when I remove the call to _fetchFBWStatus in the JS code on the router, the lime-app stops complaining, and the same happens if I replace the content of _fetchFBWStatus with something always setting lock to whatever value.

> I have this response `{"jsonrpc":"2.0","id":4,"error":{"code":-32000,"message":"Object not found"}}` to the the `lime-fbw status` call and the page shows no error.

I know how to test this just on the command line...
```
# ubus call lime-fbw status
Command failed: Not found
```

spiccinini, 2020-11-01:
> I have tested and it is working for me.
>
> So, lime-app is opening for you even without lime-fbw and ubus-lime-fbw?
Yes. Somehow we are not testing exactly the same thing. You aré using the
.js files that are in the release tarball?

Maybe it is another part of the initialization that is failing to you?

ilario, 2020-11-07:
I tested also on a second router: the fix does not work for me (I compiled the latest code using the instructions on the development page, no lime-fbw nor ubus-lime-fbw).

I checked and I can confirm that in my case the error is caused by the error handling here:
https://github.com/libremesh/lime-app/blob/v0.2.10/src/utils/app.context.js#L84-L85

I managed to compile lime-app and to deploy a locally modified version, but I know zero JS and I cannot find a way to fix this.

Can anyone else confirm if it works or if it does not? @rallep71 @AndyMcSchopf

spiccinini, 2020-11-08:
I was able to reproduce it with 19.07. I think that we are hitting a bug  in 19.07's rpcd or ubus.

Here is the request to lime-fbw, that as it is not installed is returning an error: 
[image]

And then for some reason the next request gets a size 0 response from the node. 
[image]
This size 0 response is finally rising an error (it corresponds to the upgrade info request).

In 18.06 this does not happen:
[image]

ilario, 2020-11-08:
WOW!
Amazing @spiccinini! This is a huge step forward!
@aparcar @dangowrt does this sound like a known issue to you?

AndyMcSchopf, 2020-11-09:
> I tested also on a second router: the fix does not work for me (I compiled the latest code using the instructions on the development page, no lime-fbw nor ubus-lime-fbw).
>
> I checked and I can [...]

Well I have the "mystery" that i get the erromessage when using my iPhone - but not when connected by ethernet directly to the accesspoint... 
There I can change (for example) the map - BUT I see anything on the status or the metrics page...

So I have to add this is tested on my netgear ex6130 with lime 20.09 and openwrt snapshot (today) and just another "[problem](https://lists.libremesh.org/pipermail/lime-users/2020-November/001794.html)" found....

rallep71, 2020-11-09:
@AndyMcSchopf 
Hi, I will be back for testing in the course of the week, had little time the last weeks

AndyMcSchopf, 2020-11-09:
On my ex6150v2 with openwrt 19.04 it shows the error wired and wireless (compiled without fbw).

spiccinini, 2020-11-10:
I have investigated this issue a bit. I think that the problem is related to this uhttpd commit https://git.openwrt.org/?p=project/uhttpd.git;a=commit;h=975dce23257e713e5e52eb87b194513eb81110a5 that enables keep alive in post requests. The issue is then fixed byt https://git.openwrt.org/?p=project/uhttpd.git;a=blobdiff;f=client.c;h=6233d01cd43508c6a9f15026985da21abead8cce;hp=b5736e4b00fbd2bd66784b5b551ff2051ccb36f6;hb=0f38b0370718518e66dabeaa5522f546cb459393;hpb=88ba2fa7b414f3f63e66d1708f5bda9465e0ee89 that is in the master branch but not in the 19.07 branch. I tried appying the patch and building the package and it fixes the probem.
I don't know what is the procedure to backport this fixes to the 19.07 branch.

spiccinini, 2020-11-10:
Also configuring        uhttpd with  `option http_keepalive '0'` workarrounds the issue,

spiccinini, 2020-11-13:
I've opened https://bugs.openwrt.org/index.php?do=details&task_id=3443 in hope that the next 19.07 version is realeased with the fix.

ilario, 2020-11-16:
> Also configuring uhttpd with `option http_keepalive '0'` workarrounds the issue,

Waiting for the commit to be picked in OpenWrt 19.07, do you mind applying this workaround and adding it to the list of temporarily-needed code in #809?

spiccinini, 2020-11-17:
> > Also configuring uhttpd with `option http_keepalive '0'` workarrounds the issue,
> 
> Waiting for the commit to be picked in OpenWrt 19.07, do you mind applying this workaround and adding it to the list of temporarily-needed code in #809?

I am not sure about the side effects of this.

ilario, 2020-11-17:
> I am not sure about the side effects of this.

https://forum.openwrt.org/t/massive-interface-lag-on-chrome-with-luci/56673/15

ilario, 2020-11-20:
Seems that Jow solved for 19.07 branch, so it will be included in 19.07.5 :)
https://bugs.openwrt.org/index.php?do=details&task_id=3443
https://git.openwrt.org/a448ad7490

ilario, 2020-11-22:
While we wait for 19.07.5 to be released, I edited the development page suggesting to use the openwrt-19.07 branch which already includes the fix: https://github.com/libremesh/lime-web/commit/1185edd6e1d7ef51581893cddb9f146b1a285cfb

spiccinini, 2020-11-22:
Great idea Ilario thanks

spiccinini, 2020-11-25:
With the latest web instructions (using 19.07 branch) it is working so I am closing.

closed 2020-11-25 by spiccinini