libremesh/lime-packages #767: Set new location of repository for compiled packages

ilario, 2020-09-28:
Thanks to @aparcar we have a compiled packages and network-profiles repository (see #728 and https://github.com/libremesh/network-profiles/pull/64) on https://github.com/libremesh/lime-feed

The location location of the old repository needs to be replaced, it is here:
https://github.com/libremesh/lime-packages/blob/903655831d74098ea2eba63da36d905636c6d4d0/packages/lime-system/files/etc/uci-defaults/92_add-lime-repos#L25-L31

The files on the new repository cannot be listed (currently) but can be accessed using their explicit name, for example: https://libremesh.github.io/lime-feed/master/Packages.manifest . This should not be a problem for OPKG.

aparcar, 2020-09-28:
We have to allow HTTP on the GtiHub page

aparcar, 2020-09-28:
For that we need a custom domain, e.g. repo.libremesh.org. What do others suggest?

ilario, 2020-09-28:
@spiccinini should have access to the DNS.
repo.libremesh.org is ok, I think, as currently points to the same direction as downloads.libremesh.org

ilario, 2020-09-28:
an alternative: packages.libremesh.org

spiccinini, 2020-09-29:
packages.libremesh.org is a good idea because we wont be breaking all the current links to repo.libremesh.org ordownloads.libremesh.org.

ilario, 2020-09-29:
> we wont be breaking all the current links to repo.libremesh.org or downloads.libremesh.org.

Anyway it makes very little sense to have both repo.libremesh.org and downloads.libremesh.org pointing to the same content.
All the references to repo.libremesh.org are safe to remove from LibreMesh code https://github.com/search?q=org%3Alibremesh+%22repo.libremesh.org%22&type=code I'll make a PR on this.
@aparcar proposes also another alternative: feed.libremesh.org
All the options sound equivalent, let's choose one.

aparcar, 2020-09-29:
let's use feed to not screw any existing setups

ilario, 2020-09-29:
Perfect! I set the feed.libremesh.org "Custom domain" in lime-feed repository settings.
That repository is accessible via https://libremesh.github.io/lime-feed so the DNS should point from feed.libremesh.org to libremesh.github.io correct?

ilario, 2020-10-01:
What should be the key_name and the key_content value?

spiccinini, 2020-10-02:
I've asked CodigoSur to point the feed.libremesh.org to the github.io, it should be forking now!

ilario, 2020-10-03:
> I've asked CodigoSur to point the feed.libremesh.org to the github.io, it should be forking now!

Ammazzing!
It works perfectly!
http://feed.libremesh.org/master/Packages.manifest
(there is no file listing, so that the homepage will be a 404, but that should not be a problem for OPKG)

@aparcar was asking on the Element chatroom how to get the private keys that were used for signing the packages on repo.libremesh.org in order to use them for signing feed.libremesh.org stuff. @spiccinini do you have access?

spiccinini, 2020-10-05:
I've replaced te KEY_BUILD secret in the config with the private key for the pub key:
```
untrusted comment: signed by libremesh.org key a71b3c8285abd28b
RWSnGzyChavSiyQ+vLk3x7F0NqcLa4kKyXCdriThMhO78ldHgxGljM/8
```

ilario, 2020-10-12:
Thanks!
Check out #776

PR #776 (2020-10-12): Set LibreMesh packages repository location
Thanks to @aparcar and @spiccinini now we have again a binary repository.
This sets its location in the firmware so that OPKG can use it.
Fix #767
closed 2020-10-12 by spiccinini