libremesh/lime-packages #769: search for a possebillaty to upgrade keap settings just with lime_node

der-nele, 2020-10-01:
for keap config clean
thanks.

ilario, 2020-10-03:
Check out the `lime-sysupgrade` command from lime-system package:
https://github.com/libremesh/lime-packages/blob/fa4fd5f8d3f78ace52a52b53bea87e06a3ed7827/packages/lime-system/files/usr/bin/lime-sysupgrade#L19-L22

it should do exactly what you ask for (but I never used it).
The files which will be preserved, by default are these ones:
https://github.com/libremesh/lime-packages/blob/187deb71f14c34bf3d351256865df09890e20406/packages/lime-system/files/etc/config/lime-defaults#L11

and these ones:

https://github.com/libremesh/lime-packages/blob/c4fb7118e8393b65264626cb60340f0bb0228793/packages/lime-system/files/lib/upgrade/keep.d/libremesh#L1-L3

So for having the private SSH key preserved you will have to have something like:
```
config lime system
	option keep_on_upgrade 'libremesh /etc/sysupgrade.conf /etc/dropbear/dropbear_rsa_host_key'
```
in /etc/config/lime-community (or in lime-node).

To keep the SSH key makes a lot of sense, I'll prepare a pull request with that.

I'll close the issue as I think that lime-sysupgrade is what you were looking for, but feel free to re-open it if there is more!

ilario, 2020-10-03:
Sorry, I just edited my previous comment as I got the wrong version of the lime-defaults. The SSH keys are already being preserved by default by lime-sysupgrade.

ilario, 2020-10-03:
I tested the lime-sysupgrade on a TP-Link WDR3600 and do not preserve the configuration as it should.
Seems that in the current state it can work only with the LibreRouter.
I am preparing a pull request to fix it.

spiccinini, 2020-10-03:
Ilario,.the keep on upgrade list must contain filelists and not files itself, or at least that was the design decision made by pau as far as I understand

spiccinini, 2020-10-03:
Disregard my last comment, I read your response in my email and didn't  understand it correctly.

ilario, 2020-10-03:
Oky :)
Check out #772

spiccinini, 2020-10-03:
Great! Thanks Ilario!

der-nele, 2020-10-03:
thanks for that. so my computer skills are small so i ride this and than i do just keep the lime node root@node1: lime-sysupgrade /tmp/libremeshTEST.bin [lime-node]?

ilario, 2020-10-03:
Sorry for that but lime-sysupgrade currently does not work.
We are fixing it on #772

ilario, 2020-10-14:
@der-nele the lime-sysupgrade has been fixed in the LibreMesh master branch.
This means that you'll need to update the feeds of your compilation directory (or to start a new compilation from scratch), flash the new image to your routers with the normal `sysupgrade -n /tmp/openwrt...sysupgrade.bin` and from there on you will have available the `lime-sysupgrade /tmp/openwrt...sysupgrade.bin` command.
Please confirm if it correctly solves your issue!

PR #772 (2020-10-14): Improvements to lime-sysupgrade
Fix #769 
Here goes three changes to lime-sysupgrade script from lime-system package:

1) when one of the files from a collection was missing, for example /root/.ssh/known_hosts from dropbear collection, a misleading warning was printed, for example:

`Warning: Cannot read /lib/upgrade/keep.d/dropbear file`

even if `/lib/upgrade/keep.d/dropbear` exists.

2) the checksums since OpenWrt 19.07 are SHA256 by default, so I added the output of sha256sum. I think that sha256sum was not present in OpenWrt 18.06, so I added a check for its presence.

3) in case the rout [... cut]
closed 2020-10-14 by aparcar