libremesh/lime-packages #786: FBW initial prompt: add a "Do not ask anymore" button

ilario, 2020-10-20:
Currently, in the FirstBootWizard web interface the user is welcomed with "Please configure your network" prompt.
The possible interactions are "OK" and "CANCEL".
A button like "Stop asking, I'll configure my router by hand!" is missing.
Up to now the way to do this is editing the /etc/config/lime-node  
https://github.com/libremesh/lime-packages/blob/6de76482829a774d9dec071c5d442db1c35da146/packages/lime-docs/files/www/docs/lime-example.txt#L25
setting:

```
config lime system
	option firstbootwizard_configured true
```

as commented in https://github.com/libremesh/lime-packages/issues/640#issuecomment-557199692

ilario, 2020-10-31:
https://github.com/libremesh/lime-packages/pull/792 has been merged.
If I got it correctly, once https://github.com/libremesh/lime-app/pull/278 is also merged this will be fixed.

germanferrero, 2021-02-24:
It got merged, this issue is now solved :)

closed 2021-02-24 by germanferrero