libremesh/lime-packages #787: Babeld and firewall

ilario, 2020-10-23:
`lime-proto-bmx6` has some rules to apply in case the OpenWrt firewall is installed ([here](https://github.com/libremesh/lime-packages/blob/2c4b783f68860f2542b3910fc6c46f9566b1b205/packages/lime-proto-bmx6/files/usr/lib/lua/lime/proto/bmx6.lua#L142-L155)), and I think a similar thing is needed for `lime-proto-babeld`.

On a TP-Link WDR3600, when firewall is activated I can see a:
`daemon.err babeld: send: Operation not permitted`
message in `logread` which is not present when the firewall is disabled.

Some users use FW3 (OpenWrt firewall) in the wild (for example in case they connect a LibreMesh node directly to a modem exposing the router WAN port to the internet, see #280).
@dangowrt do you see this message in your logs?

ilario, 2020-10-25:
Additionally, when firewall is enabled I see *sometimes* also this error:

`daemon.err babeld: send: Permission denied`

ilario, 2020-10-26:
The "Operation not permitted" errors gets originated from this call to sendmsg:
https://github.com/jech/babeld/blob/babeld-1.9.2/net.c#L160

ilario, 2020-10-26:
@jech do you have idea of which firewall rules are we missing?

jech, 2020-10-26:
Babel uses UDP/IPv6 with source port = destination port = 6696 and a source address that is a link-local address (is in fe80::/64).  The destination address may be either the Babel multicast address or a unicast address — Babel uses both multicast and unicast in normal operation.

It should be safe to simply allow all packets with dport=6696, since babeld itself will drop any packets that don't have a source address in fe80::/64.  In case something goes wrong, however, it might be slightly safer to only allow packets with destination port 6696 when their source address is in fe80::/64.

Note that Babel packets may be sent to both multicast and unicast addresses, so you must not match on the destination address

ilario, 2020-10-30:
Thanks @jech for the answer, superclear :)
Looking at the OpenWrt firewall configuration here https://github.com/openwrt/openwrt/blob/master/package/network/config/firewall/files/firewall.config seems to me that there is no rule blocking outgoing UDP packets, so I still don't understand why sendmsg is complaining.
Do you agree?
Is there some way to get some debug info from sendmsg or from the socket or from Babeld? I can add some code into the message.c file of Babeld and then compile it, but it's not easy to me to find out the code for printing all the fields provided to sendmsg.

dangowrt, 2020-10-30:
Never seen any of those errors when using libremesh with fw3 (ie. the only way we deploy it here unless it's only for batman-adv and doesn't involve actual layer-3 routing).
However, I've never used babeld. The rules for bmx6/7 you mentioned above are necessary for the additional (tun) interfaces created by bmx7 which would otherwise not be covered by any firewall zone. In case babeld also creates additional interfaces, you'd indeed need something similar to assign a firewall zone to them.

jech, 2020-10-31:
All Babel packets get sent by `babel_send` in `net.c`: https://github.com/jech/babeld/blob/master/net.c#L160
The simplest way is to debug is to run babeld under `strace -e sendmsg`.

I don't think there's anything special about the way `babeld` sends its packets; as @dangowrt suggests, there's probably something else interfering.

dangowrt, 2020-10-31:
I guess the easiest way to find out what is going on would be to enable logging in all firewall zones.
```
uci set firewall.@zone[1].log=1
uci set firewall.@zone[0].log=1
uci commit firewall
```
That should result in every blocked packet leaving a log line with clear hints regarding why it was blocked.
I still suspect that LibreMesh doesn't add all interfaces to firewall zones and then babel trying to sendmsg() may fail for those interfaces which haven't been added to any zone.

Another less likely guess is that it could be a problem to send on a 'closed' socket (ie. OOB). For that maybe try
```
uci set firewall.@defaults[0].drop_invalid='0'
uci commit firewall
```

If it is the `wan` interface affected, it may by the mechanism preventing leakage of apparent local traffic. Try disabling that using
```
uci set firewall.@zones[1].masq_allow_invalid='0'
uci commit firewall
```

(All the above assumes that `@zones[0]` is the LAN zone while `@zones[1]` is the WAN zone)

dangowrt, 2020-10-31:
@ilario Please share `uci show firewall` to shed more light into this.

ilario, 2020-11-02:
Thanks @jech and @dangowrt for the indications!!!!

Could it be that Babeld starts before some interfaces have an IPv6 address?
I tried moving /etc/rc.d/S70babeld to /etc/rc.d/S99babeld with no difference.

In each boot, I have a random number of these errors, from 0 to 3 approx. And this happens only at early boot, does not appear later. It could be a race condition (that maybe gets fixed automatically later?).

Do you think this could be related to #795?
@spiccinini do you see such messages?

The strace -e sendmsg worked amazingly, here is some output (selecting just the errors):

```
sendmsg(6, {msg_name={sa_family=AF_INET6, sin6_port=htons(6696), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "ff02::1:6", &sin6_addr), sin6_scope_id=if_nametoindex("eth0-1_17")}, msg_namelen=28, msg_iov=[{iov_base="*\2\0\10", iov_len=4}, {iov_base="\4\6\0\0\204\"\1\220", iov_len=8}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPERM (Operation not permitted)
sendmsg(6, {msg_name={sa_family=AF_INET6, sin6_port=htons(6696), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "ff02::1:6", &sin6_addr), sin6_scope_id=if_nametoindex("wlan1-mesh_17")}, msg_namelen=28, msg_iov=[{iov_base="*\2\0006", iov_len=4}, {iov_base="\4\6\0\0\351E\1\220\7\6\1\0\n\rh\2\6\n\0\0\302J\0\377\376\335h\2\10\f\1\0"..., iov_len=54}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPERM (Operation not permitted)
```

The current firewall UCI configuration is:

```
# uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='ACCEPT'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].mtu_fix='1'
firewall.@zone[0].network='lan' 'lm_net_br_lan_anygw_if' 'lm_net_eth0_1_batadv_if' 'lm_net_eth0_1_babeld_if' 'lm_net_wlan0_mesh_batadv_if' 'lm_net_wlan0_mesh_babeld_if' 'lm_net_wlan1_mesh_batadv_if' 'lm_net_wlan1_mesh_babeld_if' 'lm_net_eth0_batadv_if' 'lm_net_eth0_babeld_if' 'lm_net_eth0_2_batadv_if' 'lm_net_eth0_2_babeld_if'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].src_ip='fc00::/6'
firewall.@rule[3].dest_ip='fc00::/6'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Input'
firewall.@rule[5].src='wan'
firewall.@rule[5].proto='icmp'
firewall.@rule[5].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type' 'router-solicitation' 'neighbour-solicitation' 'router-advertisement' 'neighbour-advertisement'
firewall.@rule[5].limit='1000/sec'
firewall.@rule[5].family='ipv6'
firewall.@rule[5].target='ACCEPT'
firewall.@rule[6]=rule
firewall.@rule[6].name='Allow-ICMPv6-Forward'
firewall.@rule[6].src='wan'
firewall.@rule[6].dest='*'
firewall.@rule[6].proto='icmp'
firewall.@rule[6].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type'
firewall.@rule[6].limit='1000/sec'
firewall.@rule[6].family='ipv6'
firewall.@rule[6].target='ACCEPT'
firewall.@rule[7]=rule
firewall.@rule[7].name='Allow-IPSec-ESP'
firewall.@rule[7].src='wan'
firewall.@rule[7].dest='lan'
firewall.@rule[7].proto='esp'
firewall.@rule[7].target='ACCEPT'
firewall.@rule[8]=rule
firewall.@rule[8].name='Allow-ISAKMP'
firewall.@rule[8].src='wan'
firewall.@rule[8].dest='lan'
firewall.@rule[8].dest_port='500'
firewall.@rule[8].proto='udp'
firewall.@rule[8].target='ACCEPT'
firewall.@include[0]=include
firewall.@include[0].path='/etc/firewall.user'
firewall.include_firewall_lime=include
firewall.include_firewall_lime.path='/etc/firewall.lime'
```

Activating the firewall zones log option as indicated works but indicates only rejects on WAN port not exactely at the same time of the errors.

After setting `firewall.@defaults[0].drop_invalid='0'` the messages are still there.

After setting `firewall.@zone[1].masq_allow_invalid='0'` the messages are still there.

> I still suspect that LibreMesh doesn't add all interfaces to firewall zones and then babel trying to sendmsg() may fail for those interfaces which haven't been added to any zone.

Makes sense, I paste here the `ip link show` and the `/etc/config/network` together with the rest of the lime-report output:
http://sprunge.us/yChs07

ilario, 2020-11-03:
The number of errors present is really variable, sometimes there is none, this time there was:

```
sendmsg(6, {msg_name={sa_family=AF_INET6, sin6_port=htons(6696), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "ff02::1:6", &sin6_addr), sin6_scope_id=if_nametoindex("eth0-1_17")}, msg_namelen=28, msg_iov=[{iov_base="*\2\0\10", iov_len=4}, {iov_base="\4\6\0\0X\321\1\220", iov_len=8}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPERM (Operation not permitted)
sendmsg(6, {msg_name={sa_family=AF_INET6, sin6_port=htons(6696), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "ff02::1:6", &sin6_addr), sin6_scope_id=if_nametoindex("eth0-2_17")}, msg_namelen=28, msg_iov=[{iov_base="*\2\0006", iov_len=4}, {iov_base="\4\6\0\0C\23\1\220\7\6\1\0\n\rh\2\6\n\0\0\302J\0\377\376\335h\2\10\f\1\0"..., iov_len=54}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPERM (Operation not permitted)
sendmsg(6, {msg_name={sa_family=AF_INET6, sin6_port=htons(6696), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "ff02::1:6", &sin6_addr), sin6_scope_id=if_nametoindex("eth0-1_17")}, msg_namelen=28, msg_iov=[{iov_base="*\2\0\10", iov_len=4}, {iov_base="\4\6\0\0X\323\1\220", iov_len=8}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPERM (Operation not permitted)
sendmsg(6, {msg_name={sa_family=AF_INET6, sin6_port=htons(6696), sin6_flowinfo=htonl(0), inet_pton(AF_INET6, "ff02::1:6", &sin6_addr), sin6_scope_id=if_nametoindex("eth0-2_17")}, msg_namelen=28, msg_iov=[{iov_base="*\2\0\10", iov_len=4}, {iov_base="\4\6\0\0C\25\1\220", iov_len=8}], msg_iovlen=2, msg_controllen=0, msg_flags=0}, 0) = -1 EPERM (Operation not permitted)
```

If I don't start Babeld at boot but I start it later, no error is observed.

> Do you think this could be related to #795?

It is not related, I edited a line in Babeld init script to avoid this and no change.

dangowrt, 2020-11-04:
Is babeld even affected by this (ie. does it exchange routes and work as expected despite the logged errors?)
If babeld actually works, do I understand correctly that this could be a glitch caused by babeld running while (re-)starting firewall?

jech, 2020-11-04:
> Is babeld even affected by this (ie. does it exchange routes and work as
> expected despite the logged errors?)

Babel should recover.  It's merely logging a temporary failure to send its
data.

> If babeld actually works, do I understand correctly that this could be a glitch
> caused by babeld running while (re-)starting firewall?

It's okay (and expected) for UDP packets to be lost occasionally.  Babel
should deal with that gracefully.

ilario, 2020-11-06:
> Is babeld even affected by this (ie. does it exchange routes and work as expected despite the logged errors?)

I set up two routers, both with the firewall package, and the default route announced by one router reaches the other router.

> It's okay (and expected) for UDP packets to be lost occasionally.  Babel should deal with that gracefully.

Amazing, thanks for the help!

I am closing the issue, as seems that it is a transient problem getting solved automatically. Thanks!

closed 2020-11-06 by ilario