libremesh/lime-packages #788: libremap-agent errors

ilario, 2020-10-23:
In the logs from logread of my router I can see:

```
user.warn libremap-agent[2792]: unable to load plugin "location"; module 'luci.libremap.plugins.location' not found: 	no field package.preload['luci.libremap.plugins.location'] 	no file './luci/libremap/plugins/location.lua' 	no file '/usr/share/lua/luci/libremap/plugins/location.lua' 	no file '/usr/share/lua/luci/libremap/plugins/location/init.lua' 	no file '/usr/lib/lua/luci/libremap/plugins/location.lua' 	no file '/usr/lib/lua/luci/libremap/plugins/location/init.lua' 	no file './luci/libremap/plugins/location.so' 	no file '/usr/lib/lua/luci/libremap/plugins/location.so' 	no file '/usr/lib/lua/loadall.so' 	no file './luci.so' 	no file '/usr/lib/lua/luci.so' 	no file '/usr/lib/lua/loadall.so'
user.warn libremap-agent[2792]: unable to load plugin "babeld"; module 'luci.libremap.plugins.babeld' not found: 	no field package.preload['luci.libremap.plugins.babeld'] 	no file './luci/libremap/plugins/babeld.lua' 	no file '/usr/share/lua/luci/libremap/plugins/babeld.lua' 	no file '/usr/share/lua/luci/libremap/plugins/babeld/init.lua' 	no file '/usr/lib/lua/luci/libremap/plugins/babeld.lua' 	no file '/usr/lib/lua/luci/libremap/plugins/babeld/init.lua' 	no file './luci/libremap/plugins/babeld.so' 	no file '/usr/lib/lua/luci/libremap/plugins/babeld.so' 	no file '/usr/lib/lua/loadall.so' 	no file './luci.so' 	no file '/usr/lib/lua/luci.so' 	no file '/usr/lib/lua/loadall.so'
```

The `location` plugin is preset to enabled here:
https://github.com/libremesh/libremap-agent/blob/fbca90abd1e98b732f58942dc46653c567b6ada8/libremap-agent/files/etc/uci-defaults/80_libremap-agent#L25
and causes the error as the plugin is not present because it is a dependency only of [lime-map-agent](https://github.com/libremesh/lime-packages/tree/master/packages/lime-map-agent) which is not recommended in the [compilation guide](https://libremesh.org/development.html).
It should either be added as a dependency of libremap-agent or be preset to disabled (and added to the compilation guide if we want to recommend it).

The `babeld` plugin is preset to disabled and libremap-agent enables it if if detects that is actually installed here:
https://github.com/libremesh/libremap-agent/blob/fbca90abd1e98b732f58942dc46653c567b6ada8/libremap-agent/files/etc/uci-defaults/80_libremap-agent#L63-L69
but then `lime-proto-babeld` enables it forcefully here:
https://github.com/libremesh/lime-packages/blob/d81c12cdf56f0370600999adf586265c7cd7668b/packages/lime-proto-babeld/files/usr/lib/lua/lime/proto/babeld.lua#L87-L88
These two lines should be removed. If the babeld libremap plugin is actually desired, it should be included in the packages to manually select in the compilation guide.

ilario, 2020-11-08:
I addressed the second part of the described issue on #797.
Regarding the first: there are two suggested ways to solve it, which one do you prefer?

In my opinion we should not include anything libremap-related in the next release:

* [libremap repositories](https://github.com/libremap/) are abandoned since years;
* http://libremap.net [is down](https://check-host.net/check-http?host=libremap.net) (no idea since when);
* in a cascade of dependencies (`libremap-agent` => `luci-lib-libremap` => `luci-base`) it selects `luci-base` and increases a lot the size of the image (`luci-base` is also selected in a cascade by `check-date-http` package, which is suggested as optional [in the development page](https://libremesh.org/development.html#compiling_libremesh_from_source_code)).

and get rid of all these issues altogether. Currently, `libremap-agent` is selected by `ubus-lime-location` which is selected by `lime-app`.

ilario, 2020-11-10:
> * in a cascade of dependencies (`libremap-agent` => `luci-lib-libremap` => `luci-base`) it selects `luci-base` and increases a lot the size of the image (`luci-base` is also selected in a cascade by [...]

I tried to understand how ubus-lime-location works and seems that it does not require libremap-agent in order to work. The selection of libremap-agent can be suggested to users willing to update the network condition on a public map server, and can be removed as a hard dependency from ubus-lime-location.

Is this correct?

This would have the bonus to allow the generation of images with lime-app without LuCI (less dependencies smaller image!).

spiccinini, 2020-11-13:
What do you think of #805 ?

ilario, 2020-11-25:
I love #805 and if we manage to merge it before the final release that will allow users to compile much smaller images, still including lime-app.

ilario, 2020-11-30:
Fixed with #797 and #805.

closed 2020-11-30 by ilario