libremesh/lime-packages #789: Babeld error "send: Address not available"

ilario, 2020-10-23:
A user reported a cryptic Babeld error message (and I confirmed it on my router and can be seen also [here](https://antennine.noblogs.org/post/2019/11/25/valsamoggia-a-pranzo-un-rapido-test-tra-due-nuovi-nodi/)):
`daemon.err babeld: send: Address not available`
I have no idea of its severity but it is worth to inspect as it could imply some malfunctioning of the Hello sending of Babeld which would make Babeld quite useless.

This "Address not available" is mentioned here:
https://man7.org/linux/man-pages/man3/errno.3.html
and seems that its codename is EADDRNOTAVAIL.

It *could* be originating inside some of the `setsockopt` calls in Babeld:
https://github.com/jech/babeld/search?q=setsockopt
or not, I mean I have no idea on which call is causing that.
Some info on setsocksopt:
https://www.ibm.com/support/knowledgecenter/ssw_ibm_i_71/apis/ssocko.htm
https://man7.org/linux/man-pages/man2/setsockopt.2.html
https://man7.org/linux/man-pages/man7/ipv6.7.html

ilario, 2020-10-26:
The errors gets originated from this call to `sendmsg`:
https://github.com/jech/babeld/blob/babeld-1.9.2/net.c#L160

ilario, 2020-10-26:
@jech do you have any insight?

ilario, 2020-11-02:
This happens exactly twice every boot, just during boot, and does not appear anymore after boot.

jech, 2020-11-02:
> This happens exactly twice every boot, just during boot, and does not appear anymore after boot.

This means that babeld is attempting to use an interface that hasn't been
assigned a link-local address yet.  It should disappear as soon as the
kernel configures a link-local address, and is harmless.

-- Juliusz

ilario, 2020-11-03:
>> This happens exactly twice every boot, just during boot, and does not appear anymore after boot.
> This means that babeld is attempting to use an interface that hasn't been assigned a link-local address yet. It should disappear as soon as the kernel configures a link-local address, and is harmless.

Thanks for the answer!!
Indeed, while testing #787 I prepended `strace -e sendmsg -o /tmp/ciao` to babeld command in /etc/init.d/babeld and this made the `send: Address not available` errors disappear.
Likely, the execution is slowed down due to the presence of strace and the addresses are assigned before being used.
Thanks for the help!!!

closed 2020-11-03 by ilario