libremesh/lime-packages #794: Package lime-debug dependency to iputils-ping6

FreifunkUFO, 2020-10-29:
when i tried to compile libremesh i failed with openwrt-trunk.
the package "iputils-ping6" seems to be missing, so i got this error:
`u@PC:~/git2/openwrt$ time make V=99`
`WARNING: Makefile 'package/feeds/libremesh/lime-debug/Makefile' has a dependency on 'iputils-ping6', which does not exist`
`WARNING: Makefile 'package/feeds/libremesh/lime-proto-bgp/Makefile' has a dependency on 'bird4', which does not exist`
`WARNING: Makefile 'package/feeds/libremesh/lime-proto-bgp/Makefile' has a dependency on 'bird6', which does not exist `
`...`
`Collected errors:`
` * satisfy_dependencies_for: Cannot satisfy the following dependencies for lime-debug:`
` *     iputils-ping6 `

when trying it with openwrt brunch 19.07 i got only these bird-errors. also on openwrt packages-download dirs there is no iputils-ping6 any more. (f.i. https://downloads.openwrt.org/snapshots/packages/i386_pentium/base/)

so, do we have to change https://github.com/libremesh/lime-packages/blob/master/packages/lime-debug/Makefile ?

ilario, 2020-10-30:
Thanks for reporting, this is the topic of the long thread [started here on the mailing list lime-users](https://lists.libremesh.org/pipermail/lime-users/2020-October/001740.html).
The proper solution would be to create a branch for development on OpenWrt snapshot and keep the master branch for development on the latest OpenWrt release. @spiccinini 
As a temporary solution, deselect the `lime-debug` package when you compile on OpenWrt snapshot.

AndyMcSchopf, 2020-10-30:
Good to see that someone else also ran into the same problem ;-)
Hope to see a solution soon...

ilario, 2020-12-11:
Hey @FreifunkUFO and @AndyMcSchopf what about recyclimg the solution implemented by @G10h4ck in #823?

G10h4ck, 2020-12-12:
> Hey @FreifunkUFO and @AndyMcSchopf what about recyclimg the solution implemented by @G10h4ck in #823?

Tat PR address also de lime-debug dependency issue so once it is merged this issue is solved too

AndyMcSchopf, 2020-12-12:
Sounds good for me ;-)

ilario, 2021-03-24:
The merging of #823 fixed this issue but just when compiling with the buildroot method and the full git history (see #863 and https://github.com/aparcar/asu/issues/267).
So I would consider this not fixed.

ilario, 2023-02-28:
Trying to update, correct me if I am wrong please.

In OpenWrt 19.07 there is both iputils-ping and iputils-ping6:
https://downloads.openwrt.org/releases/19.07.7/packages/x86_64/base/

Since OpenWrt 21.02 iputils-ping6 disappeared, likely merged iputils-ping?
https://downloads.openwrt.org/releases/21.02.1/packages/x86_64/packages/

So, the dependencies should just be different for the upcoming 2020.2 release based on OpenWrt 19.07 and for the next 2023.1 release based on OpenWrt 22.03.

ilario, 2023-04-01:
I just observed this issue compiling on OpenWrt 22.03, so I can confirm that we need to fix this for the 2023.1 release, that will be based on OpenWrt 22.03.

G10h4ck, 2023-04-03:
I am compiling against OpenWrt main branch and it works perfectly, `iputils-ping` is installed with the option to install also the legacy symlinks `ping4` and `ping6`

ilario, 2023-04-03:
I installed `shared-state` and `lime-debug` as OPKG packages on clean OpenWrt and it did not work. I got dependencies errors.

G10h4ck, 2023-04-03:
Although it is nice to be able to install like that, we really only supports building from OpenWrt buildroot

G10h4ck, 2023-04-03:
It would be, but I am not sure we manage to get there.

ilario, 2023-04-03:
Let's continue this ImageBuilder discussion on #1000

ilario, 2023-04-04:
And there is no reason for keeping the back-compatibility thing, now that we target OpenWrt 22.03 with the master branch, no?

G10h4ck, 2023-04-05:
yeah no reason to keep retro-compatibility anymore

PR #999 (2023-04-17): Remove iputils-ping retrocompatibility with OpenWrt 19.07
Fix #794 

The `ping6retrocompat.mk` file was used for compiling with OpenWrt releases having split `iputils-ping` and `iputils-ping6` or single package `iputils-ping`.

For some reason, it did not work for me, and it is not needed anymore as we are not supporting OpenWrt 19.07 on the master branch (the OpenWrt_19.07_compatible branch has been created for that).
closed 2023-04-17 by G10h4ck