libremesh/lime-packages #795: Double entries in Babeld configuration

ilario, 2020-11-02:
On my TP-Link WDR3600 Babeld configuration looks a bit broken.
I didn't inspect why.

```
# cat /var/etc/babeld.conf
local-port 30003
interface eth0-1_17
interface eth0-1_17 wired false
interface wlan0-mesh_17
interface wlan0-mesh_17 wired false
interface wlan1-mesh_17
interface wlan1-mesh_17 wired false
interface eth0_17
interface eth0_17 wired false
interface eth0-2_17
interface eth0-2_17 wired false
 redistribute ip 2000::0/3  allow
 redistribute ip 0::0/0 le 0  allow
 redistribute ip 10.0.0.0/8  allow
 redistribute ip 172.16.0.0/12  allow
 redistribute ip 0.0.0.0/0 le 0  allow
 redistribute local  deny
 redistribute  deny
```

Seems that this doesn't happen with every router, see https://github.com/libremesh/lime-packages/pull/790#issuecomment-719665446
I tested this also removing the modifications from #790 and this still happens.
Does it happen with your routers?

ilario, 2020-11-03:
I looked into it and it happens because a line with the options gets added here:
https://github.com/openwrt-routing/packages/blob/1aacfea7b3ad97fb8c549a66fa7e7526a6419bd2/babeld/files/babeld.init#L129
and a line without options gets subsequently added here:
https://github.com/openwrt-routing/packages/blob/1aacfea7b3ad97fb8c549a66fa7e7526a6419bd2/babeld/files/babeld.init#L142

@jech does this make sense? Can we safely ignore this?

Comparing the init scripts [for master](https://github.com/openwrt-routing/packages/blob/master/babeld/files/babeld.init), [for 18.06](https://github.com/openwrt-routing/packages/blob/openwrt-18.06/babeld/files/babeld.init) and [for 19.07](https://github.com/openwrt-routing/packages/blob/openwrt-19.07/babeld/files/babeld.init) there's no difference, so I don't understand how this can not happen on every router.

ilario, 2020-11-12:
@jonglezb any insight? Thanks!

jonglezb, 2020-11-13:
I remember we already discussed this somewhere.

Does this cause an actual problem? It normally shouldn't.

ilario, 2020-11-13:
> Does this cause an actual problem? It normally shouldn't.

Thanks for the answer!
I don't know how to check if it caused a problem.

The most likely problem would be to not set the "type wireless" option (sorry, in the above paste the obsolete "wired false" option was used, this was solved in #790).
But I know no way to find out is that option was set or not.

The only way I know for getting some output from Babeld is to execute `echo dump | nc ::1 30003` (yep, 30003 port is not the standard one, I know, I have no idea why the standard one was not used in 5656f6caa8bd4d60d92e4f0a96f6f2dcdd420c32).
On an isolated node, both with duplicate lines (vanilla init script) and without duplicated lines (commenting out the line in the init script where the configuration line without options is set) I have the same identical dump:

```
# echo dump | nc ::1 30003
BABEL 1.0
version babeld-1.9.2
host LiMe-dd6802
my-id c2:4a:00:ff:fe:dd:68:02
ok
add interface eth0-1_17 up true ipv6 fe80::c24a:ff:fedd:6802 ipv4 10.13.104.2
add interface wlan0-mesh_17 up true ipv6 fe80::c24a:ff:fedd:6801 ipv4 10.13.104.2
add interface wlan1-mesh_17 up true ipv6 fe80::c24a:ff:fedd:6802 ipv4 10.13.104.2
add interface eth0_17 up true ipv6 fe80::c24a:ff:fedd:6802 ipv4 10.13.104.2
add interface eth0-2_17 up true ipv6 fe80::c24a:ff:fedd:6803 ipv4 10.13.104.2
add xroute 10.13.0.0/16-0.0.0.0/0 prefix 10.13.0.0/16 from 0.0.0.0/0 metric 0
ok
```

ilario, 2023-02-28:
Seems that the weird behavor did not cause any trouble, closing.

closed 2023-02-28 by ilario