libremesh/lime-packages #796: Conventional design of branch naming

FreifunkUFO, 2020-11-05:
Lets rename the branches to the openwrt standard used names. 
there is no need for different treatment of lime-packages, it would confuse many people.

the standard is: master-branch is for development, all openwrt releases have own branches

FreifunkUFO, 2020-11-05:
example: stuff for openwrt 19.07 is named like this:
https://github.com/freifunk/openwrt-packages/tree/openwrt-19.07
https://github.com/openwrt-routing/packages/tree/openwrt-19.07
https://github.com/openwrt/packages/tree/openwrt-19.07
https://github.com/openwrt/luci/tree/openwrt-19.07
[image]

ping @spiccinini 
https://github.com/libremesh/lime-packages/issues/749

spiccinini, 2020-11-05:
Curently we are targeting 18.06 and 19.07 and maybe master with the same codebase. 
After the release we would be targeting 19.07 and 20.xx (if openwrt does a release this year) and  master. 
Having one branch per openwrt target would be a lot of work to maintain. I still think that we can achieve compatibility with multiple openwrt versions with the same codebase in an easier way than having  one branch per openwrt version, or at least that is my opinion of the last two years experience.

ilario, 2023-02-28:
@FreifunkUFO can you please check the proposal of naming that is here?
https://github.com/libremesh/lime-packages/issues/951#issuecomment-1431132608

ilario, 2024-11-09:
No updates. So I suppose this is solved enough...?

closed 2024-11-09 by ilario