libremesh/lime-packages #800: No value in $RANDOM

AndyMcSchopf, 2020-11-10:
As stated here: [lime-users-maillinglist](https://lists.libremesh.org/pipermail/lime-users/2020-November/001802.html)
I found out, that in OpenWRT snapshot there is no value initialized for the Variable $RANDOM which is used in crontab for LiMe.

AndyMcSchopf, 2021-03-04:
I checked this point and it seems that this bug is closed in SNAPSHOT and the actual branch 21.02.

ilario, 2021-03-04:
Thanks for checking!

rallep71, 2021-03-06:
> I checked this point and it seems that this bug is closed in SNAPSHOT and the actual branch 21.02.

Question: can we include this in the current version 19.07.7 or do we have to wait until version 21.02?

ilario, 2021-03-08:
> Question: can we include this in the current version 19.07.7 or do we have to wait until version 21.02?

Eh?
Define "this" please.
Is the bug present in 19.07.7?

AndyMcSchopf, 2021-03-08:
@rallep71 : This bug was only in master, please check my mail on the mailinglist - there is/was no problem in 19.07.x ;-)

rallep71, 2021-03-08:
> > Question: can we include this in the current version 19.07.7 or do we have to wait until version 21.02?
> 
> Eh?
> Define "this" please.
> Is the bug present in 19.07.7?

Logfile, 19.07.7
Mon Mar  8 13:35:00 2021 cron.info crond[936]: USER root pid 23379 cmd ((sleep $(($RANDOM % 120)); shared-state sync dnsmasq-leases &> /dev/null)&)
Mon Mar  8 13:35:00 2021 cron.info crond[936]: USER root pid 23380 cmd ((sleep $(($RANDOM % 120)); shared-state sync babeld-hosts &> /dev/null)&)
Mon Mar  8 13:35:00 2021 cron.info crond[936]: USER root pid 23381 cmd ((sleep $(($RANDOM % 120)); shared-state sync bat-hosts &> /dev/null)&)
Mon Mar  8 13:35:00 2021 cron.info crond[936]: USER root pid 23382 cmd ((sleep $(($RANDOM % 120)); shared-state sync nodes_and_links &> /dev/null)&)
Mon Mar  8 13:35:00 2021 cron.info crond[936]: USER root pid 23383 cmd ((sleep $(($RANDOM % 120)); shared-state sync dnsmasq-hosts &> /dev/null)&)
Mon Mar  8 13:36:00 2021 cron.info crond[936]: USER root pid 23763 cmd ((for dataFile in /var/shared-state/data/* ; do shared-state bleach $(basename $dataFile .json); done &> /dev/null)&)
Mon Mar  8 13:38:00 2021 cron.info crond[936]: USER root pid 24374 cmd ((for dataFile in /var/shared-state/data/* ; do shared-state bleach $(basename $dataFile .json); done &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24440 cmd ((sleep $(($RANDOM % 120)); shared-state sync dnsmasq-leases &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24441 cmd ((sleep $(($RANDOM % 600)); check-date-http &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24442 cmd ((sleep $(($RANDOM % 120)); shared-state sync babeld-hosts &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24443 cmd ((sleep $(($RANDOM % 120)); shared-state sync bat-hosts &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24445 cmd ((sleep $(($RANDOM % 120)); shared-state sync nodes_and_links &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24446 cmd ((sleep $(($RANDOM % 120)); shared-state sync dnsmasq-hosts &> /dev/null)&)
Mon Mar  8 13:40:00 2021 cron.info crond[936]: USER root pid 24447 cmd ((for dataFile in /var/shared-state/data/* ; do shared-state bleach $(basename $dataFile .json); done &> /

spiccinini, 2021-03-08:
@rallep71 that log is normal (no error). The variables are replaced on execution, but they are loged "as they are" in the cron config.

rallep71, 2021-03-08:
> @rallep71 that log is normal (no error). The variables are replaced on execution, but they are loged "as they are" in the cron config.

Thank you for the clarification!!! Mercy

spiccinini, 2021-03-08:
Thank you and @AndyMcSchopf for the feedback and being alert of posible problems!

ilario, 2022-12-28:
On the [Matrix/Element chat](https://libremesh.org/communication.html), Pony wrote that the $RANDOM is empty, again.

This seems related to the lack of effect of the `BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT`, that gets set by a number of LibreMesh packages if they are included at compilation time in the buildroot.

This is not the only variable that can be not present when alternative installation methods are used (e.g. ImageBuilder SDK or installing packages one by one via OPKG), more of these configuration options have been listed here:
http://web.archive.org/web/20221228204719/https://listas.librerouter.org/archives/list/lime-users@lists.libremesh.org/message/AWSAUYSKEXN5GNORLIYHPJP5LIWAC5EQ/

ilario, 2022-12-28:
In my opinion we should stop using $RANDOM and use either math.random from Lua where possible (some care has to be employed for this to work, see https://stackoverflow.com/questions/21806967/lua-math-random) or some handmade pseudorandom number generator. Actually, we already have a pseudorandom generator that can be instructed to output numeric strings: 

https://github.com/libremesh/lime-packages/blob/8b54080ff47554028eec6aa9325e70ab064128af/packages/lime-system/files/usr/lib/lua/lime/utils.lua#L363-L365

But the $RANDOM variable gets used in shell scripts, so it would be more convenient to avoid calling Lua things. We could use this: https://stackoverflow.com/a/4508380

pony1k, 2022-12-29:
Turns out that `BUSYBOX_CUSTOM` has to be selected to make selection of `BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT` possible. Maybe this could be fixed by making the packages that use `$RANDOM` dependent of `BUSYBOX_CUSTOM`?
We could use `$(head -c 2 /dev/urandom | hexdump -e '2/2 "%u""\n"')` instead of `$RANDOM`. It generates numbers from 0 to 65535.

ilario, 2023-03-26:
Just proposed a package with an alternative command that can be used here:
https://github.com/openwrt/packages/pull/20728

closed 2023-04-14 by G10h4ck

ilario, 2023-04-26:
$RANDOM has been activated in OpenWrt with https://github.com/openwrt/openwrt/pull/12465 and https://github.com/openwrt/openwrt/commit/8f427f1a058dd5dcff21246a9a6d91318f55f80a, should be available in the next 23.something release.

PR #991 (2023-04-14): Adding the random-numgen command and use it for removing usage of $RANDOM
Fix #800 
Replaces the (other half of) #980 

When https://github.com/openwrt/packages/pull/20728 will be included in openwrt-22.03, the package will need to be removed from lime-packages repository.