libremesh/lime-packages #801: No IPv4 address for connected meshnodes

AndyMcSchopf, 2020-11-10:
Like described [here](https://lists.libremesh.org/pipermail/lime-users/2020-November/001800.html)
is a change in OpenWRT snapshot responsible for not starting babeld.
My actual workaround is a script which calls babeld with the command "trace" at startup.
File: /etc/init.d/S70babeld
linked to /etc/rc.d/S70babeld
with following content
```
#!/bin/ash
/etc/init.d/babeld trace
```

AndyMcSchopf, 2020-11-21:
I tested a new snapshot from yesterday - the bug appeard any more. 
Hopefully it is fixed!

ilario, 2020-11-25:
So we can close this? (and open another one for the new one you reported in the mailing list?)

AndyMcSchopf, 2020-11-27:
It looks like this Problem is solved atm. ;-)
Yes we can open a new one for the other.

AndyMcSchopf, 2020-12-03:
Problem is solved for me since a week - so it could be closed.

closed 2020-12-03 by AndyMcSchopf