libremesh/lime-packages #803: ubus-lime-metric uses /cgi-bin/hostname which belongs to FBW but do not depend from FBW

ilario, 2020-11-10:
ubus-lime-metric is looking for /cgi-bin/hostname on the other nodes:

https://github.com/libremesh/lime-packages/blob/3883b01915b4e646d685812736c4e425b4ef1d21/packages/ubus-lime-metrics/files/usr/sbin/last_internet#L17

https://github.com/libremesh/lime-packages/blob/3883b01915b4e646d685812736c4e425b4ef1d21/packages/ubus-lime-metrics/files/usr/sbin/last_internet#L28

but this file is not provided in ubus-lime-metric, instead it is provided by first-boot-wizard which is not a dependency of ubus-lime-metric, and it should not be.

ubus-lime-metric should also provide a /cgi-bin/hostname equivalent file. In order to avoid file conflicts that file can be generated with uci-defaults or simply have another name (I prefer the second option).

ilario, 2020-11-10:
And ubus-lime-metric should depend on uhttpd.

ilario, 2020-11-19:
@spiccinini opinions?
Adding a /www/cgi-bin/ubus-lime-metric-hostname to ubus-lime-metric package?

ilario, 2020-11-25:
I'm not sure it's a release blocker but for sure this will break ubus-lime-metrics with the current compilation instructions (not including FirstBootWizard).

spiccinini, 2020-11-26:
Thanks Ilario for clarifying this issue.

spiccinini, 2020-11-27:
I am proposing in #815 to move the cgi-bin/hostname into ubus-lime-utils and then make first-boot-wizard and ubus-lime-metrics depend on it.

ilario, 2020-11-27:
Great solution! It makes a lot of sense in my opinion, thanks @spiccinini!

PR #815 (2020-11-27): First boot wizard: move cgi-bin/hostname and improve dependencies
The idea is to move cgi-bin/hostname into a common package. I used ubus-lime-utils, it doesn't feel exactly right but I think it is better than what we have now.

Fixes #803
closed 2020-11-27 by spiccinini