libremesh/lime-packages #807: FBW: 403 error when fetching lime-community

ilario, 2020-11-14:
Originally, the `/etc/config/lime-community` file has `-rw-r--r--` permissions.
When creating a new network, the `/etc/config/lime-community` file has read permissions only for root: `-rw-------`.
This causes the failure in wget (403: Forbidden) on the other nodes when trying to download the lime-community file.

Before creating a new network:

```
# ls -l /etc/config/lime*
-rw-r--r--    1 root     root           196 Sep  6 16:19 /etc/config/lime-app
-rw-------    1 root     root          2300 Nov 10 12:19 /etc/config/lime-autogen
-rw-r--r--    1 root     root           209 Sep  6 16:19 /etc/config/lime-community
-rw-r--r--    1 root     root          2377 Sep  6 16:19 /etc/config/lime-defaults
-rw-r--r--    1 root     root           374 Nov 10 12:19 /etc/config/lime-node
```

after creating a new network using FirstBootWizard:

```
# ls -l /etc/config/lime*
-rw-r--r--    1 root     root           196 Sep  6 16:19 /etc/config/lime-app
-rw-------    1 root     root          2199 Nov 10 13:08 /etc/config/lime-autogen
-rw-------    1 root     root           237 Nov 10 13:08 /etc/config/lime-community
-rw-r--r--    1 root     root          2377 Sep  6 16:19 /etc/config/lime-defaults
-rw-------    1 root     root           134 Nov 10 13:08 /etc/config/lime-node
```

I am testing on top of OpenWrt 19.07.4

spiccinini, 2020-11-14:
Oh this is a really nasty bug....I don't understand how as umask by default is 0022 ...

spiccinini, 2020-11-17:
I am not able to reproduce it :( What is the output of `umask` in both routers?

ilario, 2020-11-17:
`umask` is 0022 on both routers.
What changes the permissions is `uci commit`.
Try with
```
ls -l /etc/config/lime-node
uci set lime-node.wifi.ap_name="ciao11"
uci commit
ls -l /etc/config/lime-node
```

Doesn't it happen for you?

spiccinini, 2020-11-25:
I was able to reproduce it. Somehow the default write permissions of uci changed in 19.07.

PR #814 (2020-12-01): fbw: share lime-community as cgi-bin
Sharing the lime-community file as a symlink needs read permissions and uci changes these file permissions in 19.07 when uci commit is run.
Sharing the file as a cgi-bin prevents this problems and is also more hackable. 

Fixes #807
closed 2020-12-01 by G10h4ck