libremesh/lime-packages #821: lime-report fails if siwtch is not present

G10h4ck, 2020-12-09:
Running `lime-report` on a device without ethernet switch like Ubiquity Picostation cause it's failure, the report is partial and doesn't even contain the `dmesg` output as you can see here http://sprunge.us/CJg6Zp also the exit code report error 1. @ilario can you take a look at it?

G10h4ck, 2020-12-09:
Apparently `dmesg` output is not included also on devices that have an ethernet switch as you can see here http://sprunge.us/cMg9U7 . @ilario `dmesg` output is fundamental information for debugging, please add it ASAP ;)

ilario, 2020-12-09:
> Apparently `dmesg` output is not included also on devices that have an ethernet switch as you can see here http://sprunge.us/cMg9U7 . @ilario `dmesg` output is fundamental information for debugging, please add it ASAP ;)

This was fixed in March and merged in October: https://github.com/libremesh/lime-packages/pull/697 please re-compile and re-flash. The latest 20 lines of dmesg should be printed.

> Running lime-report on a device without ethernet switch like Ubiquity Picostation cause it's failure, the report is partial and doesn't even contain the dmesg output as you can see here http://sprunge.us/CJg6Zp also the exit code report error 1. @ilario can you take a look at it?

The exit code is often 1, for example because bmx7 or bmx6 are missing.
The fact that the output stops after swconfig is not because the script exits, but its because swconfig is the last command in the old version of the lime-report you're using.
The printing of firewall rules after swconfig was also implemented in March and merged in October: 
https://github.com/libremesh/lime-packages/pull/697

G10h4ck, 2020-12-10:
`dmesg` output is quite important to debug driver problems, 20 lines seems definitely too few, if possible the whole `dmesg` should be included.

ilario, 2020-12-10:
> `dmesg` output is quite important to debug driver problems, 20 lines seems definitely too few, if possible the whole `dmesg` should be included.

Ok, there's no problem to include it all, at least on our side. I'm not sure if it can touch sprunge.us size limit...

closed 2020-12-25 by ilario