libremesh/lime-packages #825: Errors in eupgrade Makefile

spiccinini, 2020-12-10:
Reported by Jürgen on https://lists.libremesh.org/pipermail/lime-users/2020-December/001864.html
```
ERROR: please fix package/feeds/libremesh/eupgrade/Makefile - see logs/package/feeds/libremesh/eupgrade/dump.txt for details
Collecting package info: done
tmp/.config-package.in:72848:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:72848: symbol PACKAGE_profile-freifunk-80211s-sae-bmx7-batadv depends on PACKAGE_kmod-cfg80211
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:18758: symbol PACKAGE_kmod-cfg80211 is selected by PACKAGE_kmod-batman-adv
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:13280: symbol PACKAGE_kmod-batman-adv is selected by PACKAGE_lime-proto-batadv
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:73299: symbol PACKAGE_lime-proto-batadv is selected by PACKAGE_profile-freifunk-80211s-sae-bmx7-batadv
tmp/.config-package.in:141221:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:141221: symbol PACKAGE_wpad-mesh-wolfssl is selected by PACKAGE_wpad-mesh-wolfssl
configuration written to .config

  I didn't touch wolfssl at all.  

And dump.txt:

Package: eupgrade
Version: x
Depends: +libc +GCC_LIBSSP:libssp +USE_GLIBC:librt +USE_GLIBC:libpthread
Conflicts:
Menu-Depends:
Provides:
Section: lime
Category: LibreMesh
Repository: base
Title:
Maintainer:
Source:
Type: ipkg
Description:
https://github.com/libremesh/lime-packages/

@@

../../libremesh.mk:63: *** Package/eupgrade is missing the TITLE field.  Schluss.

```

ilario, 2020-12-10:
wpad-mesh-wolfssl warning is described in https://github.com/libremesh/lime-web/issues/125 and https://bugs.openwrt.org/index.php?do=details&task_id=3441

PACKAGE_profile-freifunk-80211s-sae-bmx7-batadv warning I cannot exactly remember why it was, something like that in the new network-profiles system one cannot deselect packages?
https://github.com/libremesh/network-profiles/blob/master/freifunk/80211s-sae-bmx7-batadv/Makefile
@dangowrt could you modify the network profile so that people do not see this warning? I think that the way to solve this would be to instruct people to select the network-profile AND manually select/deselect the package that cannot be selected/deselected by the network-profile. Or to resurrect [openwrt-metabuilder](https://github.com/aparcar/openwrt-metabuilder) that support deselecting things inside its example shell scripts :)

rallep71, 2020-12-11:
Hi at all, 
ive the same issue,
linux@linuxasus:/media/Data/Libremesh/openwrt$ make menuconfig
ERROR: please fix package/feeds/libremesh/eupgrade/Makefile - see logs/package/feeds/libremesh/eupgrade/dump.txt for details

PR #829 (2021-01-12): Refactor libremesh.mk and makefiles
The idea is to be able to use libremesh.mk like a module so that it can be more easily customized.

I still have to test a full build with this, but want to share the idea to get feedback.

Fixes #825
closed 2021-01-12 by spiccinini