libremesh/lime-packages #826: No Wlan Interface!

rallep71, 2020-12-12:
Hello all,

there has been a lot going on here, now I wanted to create a new image for TP Link Archer C50 V3.
I went through the instructions step by step "https://libremesh.org/development.html".
But unfortunately there is no wifi interface to see!

It's not the first time I've done this!
Have I forgotten something?

[image]

ilario, 2020-12-12:
Do you have lime-report command on that router?
Please share its output!
For example with `lime-report | sprunge` and share here the resulting link (sprunge works only if the router is connected to the internet).

And please check if any wpad package is installed:
`opkg list-installed | grep wpad`
some times the wpad-mesh-wolfssl deselects automatically, see https://github.com/libremesh/lime-web/issues/125
if no wpad package is selected, this would explain why the wireless is not working at all.

rallep71, 2020-12-13:
Thank you for the answer,

I was already in despair after I have compiled a new firmware for hours and it did not work, then again and again... I deleted the folder yesterday evening out of sheer rage :D :D :D. Now I'll do it again....:D

In the meantime I installed the update from Openwrt 19.07.5 to see that the wifi works :D .

That's why I can't make lime-report | sprunge at the moment, the router is also not connected to the internet at the moment.

ilario, 2020-12-13:
Ok!
Please select lime-report or lime-debug in the menuconfig :)
There's no need to have the router connected to the internet: you can run lime-report and then copy and paste all the output!

rallep71, 2020-12-13:
it's funny, when i select profile-libremesh-default and profile-libremesh-encrypt-11s under network profile, then wpad-mesh-openssl is set under network. then the wifi works again, logically.

what i also noticed, previously in Base system → firewall switched off is also set again..hm

I had also read somewhere in the documentation to enable batman-adv-auto-gw-mode on every router, normally it is not enabled.

what about bmx6-auto-gw-mode? do i need both or none? questions upon questions that i can't figure out.

ilario, 2020-12-13:
> it's funny, when i select profile-libremesh-default

This profile do not add anything on top o the default configuration. Maybe to make it useful we should add as dependencies the packages suggested in the development page...
You can find it here:
https://github.com/libremesh/network-profiles/tree/master/libremesh

> and profile-libremesh-encrypt-11s under network profile,

Do not select more than one network profile at once, weird things can happen.

> then wpad-mesh-openssl is set under network. 

Yep, because that profile selects wpad-mesh-wolfssl

> then the wifi works again, logically.

Mh, this makes me suspect that no wpad-* package was selected before, check out libremesh/lime-web#125

> what i also noticed, previously in Base system → firewall switched off is also set again..hm

Weird.

> I had also read somewhere in the documentation to enable batman-adv-auto-gw-mode on every router, normally it is not enabled.

Can you find where you read that?
I have no idea in which cases one should use batman-adv-auto-gw-mode.

> what about bmx6-auto-gw-mode? do i need both or none? questions upon questions that i can't figure out.

The most tested routing protocols for LibreMesh lately is the combination of Batman-adv+Babeld, so this package should not be useful.
The Batman-adv+BMX6 combination should work, and that case you would want to have bmx6-auto-gw-mode.
For Babeld such a package does not exist yet, see https://github.com/libremesh/lime-packages/issues/93

rallep71, 2020-12-14:
Thanks Ilario for this Answer,

I'm going to think about it and test a few things, but it's going to take a while.

It would be great if something like this were in the documentation

ilario, 2020-12-14:
> It would be great if something like this were in the documentation

Yep!
Please consider contributing in the documentation! <3

rallep71, 2020-12-20:
Hallo ilario, 
I have now done some tests and it works as it should. 
I strictly followed the instructions in the documentation.

I found one error, wpad-basic is marked as <M>, wpad-mesh-openssl <*>.
<M> deselect, save and exit, then menuconfig an select wpad-mesh-openssl <*> safe and exit, then it works :)

next:
config lime 'wifi'
        option ieee80211s_encryption 'psk2-aes'
should be changed to 
config lime 'wifi'
        option ieee80211s_encryption 'sae'.
do not bring a red error message in Luci in mesh encryption ;)

rallep71, 2020-12-20:
I have one more question:
About the custom configuration file.

I have done the following

cd openwrt
mkdir -p files/etc/config/
touch files/etc/config/lime-node (should only be used for encrypting AP's and mesh)

config lime 'system

config lime 'network

config lime 'wifi
        option ieee80211s_encryption 'sae'
        option ieee80211s_key 'hjzutredr456!#'

this would be an example to encrypt mesh as standard, if it is correct I don't know now,

what would I have to enter for the APs in order to deliver them encrypted? 

when i have everything up and running like this, i would like to make a youtube video on my channel, because i think it would be helpful for a large number of people.

because there's not much really on the topic of mesh, how do I create one myself, except very technical stuff.

ilario, 2020-12-20:
> option ieee80211s_encryption 'psk2-aes'
> should be changed to
> config lime 'wifi'
> option ieee80211s_encryption 'sae'.
> do not bring a red error message in Luci in mesh encryption ;)

Ah ehm I have no idea of the psk2-aes vs sae difference. Opinions @aparcar @dangowrt ? Which one should we suggest to use?

> cd openwrt
> mkdir -p files/etc/config/
> touch files/etc/config/lime-node (should only be used for encrypting AP's and mesh)
>
> config lime 'system
>
> config lime 'network
>
> config lime 'wifi
> option [...]

Looks good to me, I think it should work.

> what would I have to enter for the APs in order to deliver them encrypted?

Check out /docs/lime-example.txt or http://thisnode.info/lime-example.txt or https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/www/docs/lime-example.txt 

https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/www/docs/lime-example.txt#L89-L93

you need ap_encryption, ap_key, apname_encryption, apname_key.

> when i have everything up and running like this, i would like to make a youtube video on my channel, because i think it would be helpful for a large number of people.
> 
> because there's not much really on the topic of mesh, how do I create one myself, except very technical stuff.

This would be awesome!!
Please share also on the [lime-users mailing list](https://lists.libremesh.org/mailman/listinfo/lime-users) once you have it :D

> One more question, I have a Freeradius server in my network. Is it possible to authenticate on the AP's with Wpa2-EAP? Or is that not possible with the Libre network design?

No idea :) I never configured it neither on OpenWrt nor on LibreMesh. Please try and let us know!

rallep71, 2020-12-20:
Hi Ilario,
this config works good, flash the firmeware, and than restart again to bring up the 5ghz network, i dont know why, but ist works now.

config lime 'wifi'
    #option country 'DE'
    option ap_key 'xxxxx' 
    option ap_encryption 'psk2' 
    option apname_key 'xxxxx'' 
    option apname_encryption 'psk2'
    option ieee80211s_encryption 'sae'  
    option ieee80211s_key 'libremesh'  
 
question about distance 2,4ghz network and 5ghz network, 
100 meter 2,4 and 1000 meter 5ghz is the default config, performance? ive reduced 1000 meters in 5ghz network to 200 meters, 

it should only work in my house and the plot of land, approx. 1000 qm.,

rallep71, 2020-12-20:
Another question,
By default, ubus-lime-batman-adv is not selected.

should this be selected or not? if yes, why, and if no, why? :D

see picture

[image]

rallep71, 2020-12-22:
I switch back mesh encrypttion to option ieee80211s_encryption 'psk2+aes, 'some routers may have problems with this.

now i see the following on my router tp wdr4300 v1.

wlan1-mesh: MESH-SAE-AUTH-FAILURE addr=50:d4:f7:89:69:7d
wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=50:d4:f7:89:69:7e
wlan1-mesh: MESH-SAE-AUTH-BLOCKED addr=50:d4:f7:89:69:7d duration=300
wlan0-mesh: MESH-SAE-AUTH-BLOCKED addr=50:d4:f7:89:69:7e duration=300

I have read that some wifi drivers cannot establish multiple encrypted mesh connections at the same time.

I don't have such messages with my two tp link archer c50 v3, only with wdr4300,

is there a solution for this?

closed 2020-12-22 by rallep71