libremesh/lime-packages #838: WARNING: Makefile 'package/feeds/libremesh/lime-map-agent/

rallep71, 2021-01-01:
I am still testing, I have seen that something has changed on the documentation page:
https://libremesh.org/development.html
cat << EOF >> feeds.conf

src-git libremesh https://github.com/libremesh/lime-packages.git;master
src-git profiles https://github.com/libremesh/network-profiles.git
EOF

now I get this warning, can I ignore it? I don't need lime-map.

make -j$(nproc)
WARNING: Makefile 'package/feeds/libremesh/lime-map-agent/Makefile' has a dependency on 'libremap-agent', which does not exist
WARNING: Makefile 'package/feeds/libremesh/lime-map-agent/Makefile' has a dependency on 'luci-lib-libremap-location', which does not exist
WARNING: Makefile 'package/feeds/libremesh/lime-map-agent/Makefile' has a dependency on 'luci-lib-libremap-system', which does not exist
WARNING: Makefile 'package/feeds/libremesh/lime-map-agent/Makefile' has a dependency on 'luci-lib-libremap-wireless', which does not exist
WARNING: Makefile 'package/feeds/libremesh/luci-app-lime-location/Makefile' has a dependency on 'libremap-agent', which does not exist

ilario, 2021-01-02:
This happened here: https://github.com/libremesh/lime-web/commit/0bfdc9e90419b18f92f01ff9b1b0d194c72bf0a8 inside this PR: https://github.com/libremesh/lime-web/pull/132

The possible solutions are:

* ignore the warning (the warning is harmless but pleeenty of people will complain about that, so I think we cannot ignore the warning)
* revert https://github.com/libremesh/lime-web/commit/0bfdc9e90419b18f92f01ff9b1b0d194c72bf0a8 adding back the feed
* delete luci-app-lime-location (based on the old LuCI interface) and lime-map-agent (which is a metapackage)
* move both packages to the abandoned repository of libremap-agent: https://github.com/libremesh/libremap-agent/

@spiccinini what's your opinion?

rallep71, 2021-01-02:
Thanks for the answer, the network is also running like this.

But where I still have problems is in my other post with wpa_supplicant[2410]: wlan1-mesh: MESH-SAE-AUTH-FAILURE, I have read a little about it, there is not much, could be due to the hardware of the wdr4300.

rallep71, 2021-01-26:
Hello, problem solved.
mesh compiled with psk2 + aes +openssl openwrt 19.07.6

ilario, 2021-05-20:
The warning is still there, right?
So I re-open this issue.
My favorite solution would be to move those two packages to the libremap repository.

ilario, 2023-02-27:
The package that cause this issue has been removed in #943

closed 2023-02-27 by ilario