libremesh/lime-packages #842: Openwrt 19.07.06 dnsmasq error, babald error usw..

rallep71, 2021-01-24:
hello friends,

I have tested the new release of openwrt, somehow nothing works together anymore.
I would like to ask you to verify this. Maybe a user will test it again.

Sun Jan 24 14:55:16 2021 daemon.err dnsmasq[1818]: failed to send packet: Network unreachable
Sun Jan 24 14:55:17 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=b0:4e:26:45:63:ac
Sun Jan 24 14:55:17 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=64:70:02:a2:fd:23
Sun Jan 24 14:55:36 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=b0:4e:26:45:63:ac
Sun Jan 24 14:55:37 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=64:70:02:a2:fd:23
Sun Jan 24 14:55:43 2021 daemon.err dnsmasq[1818]: failed to send packet: Address family not supported by protocol
Sun Jan 24 14:55:45 2021 daemon.err dnsmasq[1818]: failed to send packet: Network unreachable
Sun Jan 24 14:55:48 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=64:70:02:a2:fd:23
Sun Jan 24 14:55:54 2021 daemon.err dnsmasq[1818]: failed to send packet: Address family not supported by protocol
Sun Jan 24 14:55:56 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=b0:4e:26:45:63:ac
Sun Jan 24 14:56:00 2021 cron.info crond[954]: USER root pid 7623 cmd ((for dataFile in /var/shared-state/data/* ; do shared-state bleach $(basename $dataFile .json); done &> /dev/null)&)
Sun Jan 24 14:56:02 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: mesh plink with 64:70:02:a2:fd:23 established
Sun Jan 24 14:56:02 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-PEER-CONNECTED 64:70:02:a2:fd:23
Sun Jan 24 14:56:14 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-FAILURE addr=b0:4e:26:45:63:ac
Sun Jan 24 14:56:14 2021 daemon.notice wpa_supplicant[4970]: wlan0-mesh: MESH-SAE-AUTH-BLOCKED addr=b0:4e:26:45:63:ac duration=300
Sun Jan 24 14:56:18 2021 daemon.err babeld[1334]: Late hello: bufferbloated neighbor fe80::b24e:26ff:fe45:63a

spiccinini, 2021-01-24:
Hi! Is this related to #837 ? I never usted SAE .
I think that maybe there are wifi firmware related issues around this configuration and that explains why on some devices it work and in some it doesn't?

rallep71, 2021-01-26:
Hello, problem solved.
mesh compiled with psk2 + aes +openssl openwrt 19.07.6

closed 2021-01-26 by rallep71

djStolen, 2021-06-06:
> Hello, problem solved.
> mesh compiled with psk2 + aes +openssl openwrt 19.07.6

Hi all, I tried to setup with psk2+aes but either, I am doing something wrong or it is not allowing me to setup psk2+aes as encryption for mesh.

Could u pls share your /config/ file where u setup this?