libremesh/lime-packages #848: FBW: make sure that the user knows the community's shared password

ilario, 2021-02-28:
When joining a community using FirstBootWizard, all the community settings are applied, including the shared password.
A user could join a community without knowing its shared password, and lose access to the router.
Before applying the new settings, FBW should ask the user the password and check it against the hash received from the remote node.

germanferrero, 2021-03-11:
Thanks Ilario for bringing this topic, I tell a little bit about the context or mindset with which we decided to implement the "shared password":

It is related to the governance scheme of a mesh network that we want to promote.
Usually, in our territory: Cordoba, Argentina, what happens is that the network is managed by a group of hackers or technical people who have ssh keys and access to the nodes of a network is guaranteed for this group of people by putting their public key in each node of the network.

This scheme makes it very difficult for people who do not have ssh keys to participate in the network administrative tasks available in the LimeApp: Firmware Upgrade so far and with more to come.

The idea of having a shared password on the network, is that whoever knows that password can have access to all the nodes on the network, as we understand that it is necessary to make changes at the network level, and at the same time it is technically accessible for anyone to gain this access, by simply knowing the password.

What happens then, is that when you join your node to a libremesh network, you accept that this is the governance scheme.
Perhaps for many of the non-technical people who join a community network, the technical governance of the network is something that they discover in a later instance after adding their node to the network through firstbootwizard.
So asking them to know a priori the shared network administration password may be an unwanted limitation.
And displaying a legend like "By joining this node you accept this governance scheme..." may mean nothing to them.

On the other hand, if you have technical knowledge, you can recover your node physically or opt out of using firstbootwizard.

Maybe we can find a solution that will work for all kinds of technical and non-technical users ☺.

ilario, 2023-02-27:
Thanks @germanferrero for the info and sorry for answering so late, I fear I missed your message.

> What happens then, is that when you join your node to a libremesh network, you accept that this is the governance scheme.

This is also an interesting topic but is not what I was bringing up.

> Perhaps for many of the non-technical people who join a community network, the technical governance of the network is something that they discover in a later instance after adding their node to the [...]

Mh?
From what you explain, they will need the password, so why not making sure they know it already?

> And displaying a legend like "By joining this node you accept this governance scheme..." may mean nothing to them.

Well, if you don't make sure they know the password, at least a warning is needed, telling them that they are going to lose access to their node if they don't know the shared password.

> On the other hand, if you have technical knowledge, you can recover your node physically

This is not always easy, especially if the node is on a roof.

> or opt out of using firstbootwizard.

Yes, in my opinion we need to make clear what implies using FBW.

> Maybe we can find a solution that will work for all kinds of technical and non-technical users relaxed.

It is not a matter of technical or non-technical, it is a matter of not loosing access to your node. This is important both for technical and non-technical people.