libremesh/lime-packages #853: ethernet bug

der-nele, 2021-03-08:
not sure if it is an libremesh problem
i admin a mesh in leipzig whis bmx7 and batman and OpenWrt 19.07.6

the bug comes ca. all second month /also wihs older versions

all ethernet connections in the lokal-mesh one of 4 ssid's are break dowm.
the mesh ist bild up out of ca 10x tplink 3600 and one loco-m (behinde the setra-5ghz are more ssid's)

______=kabel/mesh
&&&= wifi/mesh

10.182.107.83 'kh-setra-5ghz' wifi 153
                           |
               10.182.107.205 'kh-setra' wifi off________ ____kh-setra2 wifi 13 u 48
                       |                            |			&
10.182.107.92 'kh-wald' wifi 5 u 40                 |			&
         |          |                               |			&
         |          |                               |			&
10.182.107.194 'kh-bad' wifi 9 u 44                 |			&
                    |                               |			&
                    |                               |			&
        10.182.107.244 'kh-garten1' wifi 1 u 36     |                  &  
                        §                          |_________________&_____
                        §                                           &      |
                        §         10.182.107.226 'kh-garten2' wifi 13 u 48 |
                        §                                                  |
   10.182.107.104 'kh-terasse' wifi 1 u 36          _______________________|   
                                                    |
                                                    |                                       
                                                    |
        §§§§§§§§§§§§§§§§§§§§ 10.182.107.21 'kh-saloon' wifi 9 u 44 
        §                      §                           |
        §                      §                           |
        §           10.182.107.210 'kh-tor' wifi 9 u 44     |
        §                      §                           |
        §                      §             10.182.107.136 'kh-tor2' wifi 5 u 40
10.182.107.209 'kh-bauplatte' wifi 9 u 44
                               §
                               §
                      10.126.42.48 'kh-jenka'wifi 9 u 44

my problem is that the routers are isolatet if the ethernet gos down and i have to shout down the hole area from
electricity ore walk to all routers and make on/off  

thanks for interest
nele
ps: sorry for the bad spelling. is not my best skill.

der-nele, 2021-03-08:
[image]

der-nele, 2021-03-08:
the map and 2 logreads
[image]

[2logreads.txt](https://github.com/libremesh/lime-packages/files/6100926/2logreads.txt)

der-nele, 2021-03-14:
it happend again. / i think there is a connection between bmx7 and a split of one ssid.
we have often that the fuse from elektricety came. so sometime a node in the middle turn of. i think this can trigger the bug. we had this also when we did update with a new communety file  =(new ssid) as we update one node in the middle before the other notes were new. the same happens.  wifi mesh= yes but no eternet at any node in the ssid
[ethernet-bug.txt](https://github.com/libremesh/lime-packages/files/6136319/ethernet-bug.txt)

feel free to correct or ask questions for better understanding...

spiccinini, 2021-03-26:
Hi! Sorry for the delay, I've never used bmx7 myself. Do you think it is a bmx7 issue? Maybe it would be a good idea to report it to https://github.com/bmx-routing/bmx7/issues

Or do you think it is a libremesh issue? 

I don't understand exactly, the issue is that when the ethernet goes down and then back online the mesh is split and does not rejoin and you have to unplug the ethernet cable and replug it manualy?

der-nele, 2021-03-29:
jes  can be bmx7 

the ethernet interfaces fron every node are of at the same time and do not com back. althow no ping ff02::1%br-lan

ilario, 2023-02-27:
@der-nele is this issue still happening?
We are planning to not support BMX7 in the next release (see #977), as we feel that its support in LibreMesh is not active. Do you agree or it is working well for you?

der-nele, 2023-03-04:
@ilario 
yes it still happens and i do not now how. but...
for me it would be nice to have one more update with bmx7 just to not be forced for such and learn about an other firmware.

der-nele, 2023-03-04:
i already call out for helb whis an other firmware.

ilario, 2024-11-09:
Did you find a solution @der-nele ?

You can try using [deferable-reboot](https://github.com/libremesh/lime-packages/tree/master/packages/deferable-reboot) package that reboots the routers periodically. You may want to add something like this script that delays the reboot until when some check fails:
https://github.com/libremesh/lime-packages/issues/1023

As you are connecting LibreMesh routers by cable, you should really configure that ethernet port for being out of the br-lan bridge.
With DSA (modern routers with modern OpenWrt), you can usually easily configure the ethernet ports one by one. Read here:
https://github.com/libremesh/lime-packages/issues/1121
and apply to those ports this interface-specific configuration:
https://github.com/libremesh/lime-packages/blob/6642f373908d67c9cfc9d8f360843e9c528a8b94/packages/lime-docs/files/www/docs/lime-example.txt#L231-L235

If you just have something like eth0.1 and it includes all of the LAN ports, you have two options:
1) configure all of the ports of that router for mesh-only (this is a problem if you have both a libremesh node and a cabled clients connected by cable on the same libremesh router).
2) find out how to split the eth0.1 interface and then configure them separately. I personally never did it.

Sometimes the speed autonegotiation on the cable gets crazy. It could change from fast ethernet to gigabit even if gigabit does not work. In that case you will need to include an ethtool command somewhere setting the speed to fast ethernet (100M) and disabling the autonegotiation.