libremesh/lime-packages #862: Shared state publishing twice babeld_hosts and bat_hosts on boot

germanferrero, 2021-03-18:
Both:
babeld_hosts at: https://github.com/libremesh/lime-packages/blob/aa55eb84d8c02d7b23949ccc24f1b7c030d63daa/packages/shared-state-babeld_hosts/files/etc/hotplug.d/initd/shared-state-babeld_hosts#L6

and
bat_hosts at: https://github.com/libremesh/lime-packages/blob/aa55eb84d8c02d7b23949ccc24f1b7c030d63daa/packages/shared-state-bat_hosts/files/etc/hotplug.d/iface/shared-state-bat_hosts.sh#L2

Are scheduled to publish on boot (and interface up).

And also at:
https://github.com/libremesh/lime-packages/blob/aa55eb84d8c02d7b23949ccc24f1b7c030d63daa/packages/shared-state/files/etc/init.d/shared-state#L6
Shared State schedules all publishers to publish on boot, so we are doing it twice for these two, overloading the boot process.

predators46, 2021-03-18:
No found depedensi tmate

https://github.com/libremesh/lime-packages/blob/aa55eb84d8c02d7b23949ccc24f1b7c030d63daa/packages/ubus-tmate/Makefile#L17

ilario, 2023-02-24:
So, which is ok to remove? @spiccinini do you have idea?
It would be easier to remove the hotplug.d.