libremesh/lime-packages #863: Mention to a specific commit breaks builds without full git history

ilario, 2021-03-19:
Issue reported by @dangowrt in this comment:

https://github.com/libremesh/lime-packages/commit/ad8d1a3dbaa52362d6859e720aec08cb8c180a51#commitcomment-48372702

It refers to this code line:

https://github.com/libremesh/lime-packages/blob/ad8d1a3dbaa52362d6859e720aec08cb8c180a51/ping6retrocompat.mk#L17

dangowrt, 2021-04-05:
Is anyone working on resolving this?

spiccinini, 2021-04-05:
> Is anyone working on resolving this?

Hi! Not myself, if you have a patch please share :) 
Also I think that we may drop 18.06 support if it simplifies things (I don't know if it does in this case).

ilario, 2021-04-05:
OpenWrt 18.06 support should for sure be abandoned from master branch as it is in End-of-Support.

closed 2021-04-07 by dangowrt