libremesh/lime-packages #864: servers within a network with static ip might be overwritten?

nicopace, 2021-03-22:
While having a conversation with @tbdinesh this issue came about:
> We are thinking of using 1 pi for dns resolution and run bind on it. How to give a static ip on mesh for this name server?

this made me think how we manage the computers that are fixed in our networks.

One way of approaching this is to add the fixed ip and host to /etc/hosts, but you would have to maintain it yourself.

The other way is by not configuring it at all... i think that when the dhcp does a lease, it asks if that ip has been used (so it doesn't step onto anyone else)... but not sure if it happens on all interfaces (so bat0 is also included, ie the whole mesh), or not...

Anyway, I believe this should be a typical usecase that we should consider.

ilario, 2021-03-22:
One possibility is to specify an IP range for nodes autoconfiguration (with the /16/17 notation) and another for DHCP  leaving out a range available for the static IPs.

https://github.com/libremesh/lime-packages/blob/2b7226a8d249553e5e7fe3ab8e49dce2566bd318/packages/lime-docs/files/www/docs/lime-example.txt#L33-L35

nicopace, 2021-03-23:
Yes indeed, you can.
I would like to:
1. understand how it behaves when a device connects to the network and has a fixed IP address
2. consider this usecase for the non-technical or semi-technical crew: ie. someone that knows how to fix an IP address in its computer, but not in the whole mesh.

Also, I am interested in ways to approach this that don't require to reflash the whole mesh with a new set of configurations, or if needed to create the necessary changes in order for this not to be required, as it is something that belongs to the runtime of the network, and it happens in relation with a different layer, so it should expect not to be hardwired on the firmware.... right? 

eager to hear the rest here...

germanferrero, 2021-03-26:
If I got this right, I think this may somehow tackle number 2 as it does not need to be configured in the whole mesh manually:

At the node in which you have your pi / computer that needs a fixed connected add the following:

```
# /etc/lime-assets/node/my-server-static-lease.conf

dhcp-host=<MAC>,<IP>,<HOSTNAME>,infinite
```

```
# /etc/config/lime-node

(...)
config copy_asset my-server-static-lease
   option asset 'node/my-server-static-lease.conf'
   option dst '/etc/dnsmasq.d'
```

From https://github.com/libremesh/lime-packages/commit/d632accffbb39ffb7be4444387f247f755c826f7
this will make shared-state spread both the lease and the host to the network.

By using copy_asset at lime-node, the configuration will be persisted through upgrades making maintenance easier.

nicopace, 2021-03-30:
Thanks for this @germanferrero !
I think it is the easiest.
Though the best would be for the mesh to remember leases and hostnames (so users don't have to fiddle with IP Addresses management), this is a fair tradeoff.

brunovianna, 2021-04-19:
Im trying the suggestion by German, and I'm getting the following error when running lime-config after the changes:

```
/usr/bin/lua: /usr/lib/lua/lime/config.lua:122: attempt to index local 'high_pt' (a nil value)
stack traceback:
	/usr/lib/lua/lime/config.lua:122: in function 'uci_merge_files'
	/usr/lib/lua/lime/config.lua:169: in function 'uci_autogen'
	/usr/lib/lua/lime/config.lua:195: in function 'main'
	/usr/bin/lime-config:55: in main chunk
	[C]: ?
```

the files were edited like this:
/etc/lime-assets/node/static-lease.conf:

```dhcp-host=01:33:3:33:33:33:33,10.7.7.7,name,infinite```

 /etc/config/lime-node:
```config lime 'system'
        option hostname 'name'

config lime 'network'

config lime 'wifi'

config copy_asset static-lease
   option asset 'node/static-lease.conf'
   option dst '/etc/dnsmasq.d'
```

spiccinini, 2021-04-19:
Hi Bruno! I think that the problem is that UCI does not support the middle `-` in the name (static-lease) .... I believe it has to be `'static-lease'` or `static_lease`, can you try that?

ilario, 2023-02-24:
Even after years this is still useful, se need to confirm this work and document it.
@brunovianna did this work?