libremesh/lime-packages #866: lime.network: mesh_ifaces to return up&running ifaces

germanferrero, 2021-03-23:
Currently it retrieves mesh and adhoc interface names from the uci-config. But, in some cases this interfaces cannot be setup and it is not available, let's say the case in which the radio hardware is missing. Currently if this is the case the LimeApp will show the interface in the align page without handling it correctly (i.e, it stucks in a loading message state).
I think that returning only mesh and adhoc interfaces which are under `/sys/class/net/` would be more useful, for this and other use cases.

ilario, 2022-11-18:
Where should be done, this change?
In this file? https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/network.lua

ilario, 2023-02-24:
Can anyone confirm this is a real problem?
Can the hardware be different from what OpenWrt thinks?
In this case, is it going to appear in /etc/config/network?

ilario, 2023-02-27:
As @germanferrero is not actively developing for LibreMesh right now, @spiccinini do you think you can manage this?

ilario, 2024-11-08:
Wooosh dead issue since 2021 is going to be closed.

closed 2024-11-08 by ilario