libremesh/lime-packages #867: List network nodes service

spiccinini, 2021-03-26:
A list of nodes that belong to the network with their current status of being connected/disconnected and other info (like the firmware version, board, etc) is needed for multiple use cases like:

* List all the nodes of the network that its firmware version is outdated.
* List which nodes are not online but they should (they don't have power, or the antenna is misaligned, etc). 
* Have a table with the nodes and its boards. These may help to maintain
* Get a list of the online and offline nodes so that a config upgrade that needs to be performed in all the nodes at the same time can take place when all the nodes are online.
* Many many more community features

A list like this is currently maintained by hand (in spread sheets, text files, etc) in many communities. To provide these features by software we need a way to get these lists programaticaly and in a way that these information is not lost when the mesh is powered off. 

Requirements:
* The database of the list of nodes have to be populated without user/community intervention
* The database must be persistent
* The community should be able to mark a node as not participating anymore (for example people moved, or the name of the node changed, etc). When retrieving the nodes the nodes markes ad not participating are not listed (at least by default). This change should be able to be performed from any node.
* At least an attribute of the node is to be connected or disconnected (like reachable / unreachable). More attributes may be added at later.

Idea of implementation:
Use shared state in two ways:
* a multi-writer (aka multi-master) and persistent mode to store the "non volatile information": hostname, firmware version, board, etc. All the information is stored in all the nodes using /etc as storage (an not /tmp)
* use the current shared-state mode  (single-writer and non persistent) to store the "volatile information": the connection status.

G10h4ck, 2021-04-24:
I don't understand why the multi-writer is needed for the node list, can't each node publish it's own perspective and then the perspectives melded together on visualization to create a global view of the network?

spiccinini, 2021-04-27:
> I don't understand why the multi-writer is needed for the node list, can't each node publish it's own perspective and then the perspectives melded together on visualization to create a global view of the network?

Yes it is possible to persist all the perspectives of all the nodes in each node and then perform the meld in the visualization of the data (we would have to store all because if one node says that another left the network then that will be needed also stored in each node). 
In my point of view it is better to address this directly in the data layer instead of doing that above, not only for the bandwidth and storage savings (that may be negligeable) but because the system is easier to understand and matches better this use case: we want that each node has the same view of the system, if they have a different view is because it did not converge yet.
What do you think?

germanferrero, 2021-04-27:
I also think it is possible but harder to understand, since information like the nodes which left the network will be spread all over the different perspectives of each node. If we need to implement reversible changes, like "that node came back to the network", the meld would be even harder.

In the original proposal nodes converge to a unique network-nodes state, by using granular keys, one for each node and timestamp or counter + nuance as merge strategy. Which I found easier to reason about.

G10h4ck, 2021-04-29:
The problem is that the most reasonable strategy to merge the data in the most possible conflict free and distributed way is intimately tied with the semantic/nature of the data, so `shared-state` that has been developed to provide an as much as possible generic solution does not handle this because there is no **generic** way to handle it reasonably.
So the proposed strategy and optimization both in term of performance and reasoning load on the developers probably makes perfect sense for the type of data needed by network nodes service and in other cases too, but will break the conflict free distributed replication with other types of data.
So if we want to handle this kind of optimizations we should provide a way that it can be specified for each kind of data, and eventually a way to share easily those strategies between different data types if it make sense.

G10h4ck, 2021-04-29:
Can the data specific merging strategy optimizations be condensed in one or a few functions called back by `shared-state` in specific stage of it's normal operations?
In this case we could provide default implementations that doesn't change the `shared-stade` default behaviour, when a data type can provide better strategy/optimization it should just provide those functions and `shared-state` would hook on those

PR #873 (2021-07-08): Add shared state network nodes
This PR adds an rpcd/ubus service that provides the list of the nodes of the network and the ability to remove one node from the list. This uses the shared-state multiwriter as main storage so it should be easiear to first review and merge the shared-state changes of PR #872 .

Most of the code was done by @germanferrero , great work! 

Fixes #867

Implementation details (see rationale in the related issue #867):

* Use the new shared-state-multiwriter:
    * The keys are the hostnames of the nodes
    * Each node inserts its own key in the database
    * Any nod [... cut]
closed 2021-07-08 by spiccinini