libremesh/lime-packages #875: Add more info at mesh beacons.

germanferrero, 2021-04-22:
It would be great to have the opportunity to add more information in mesh beacons.

Use case: You are trying to recognize an specific node while doing frequency scan and you get something like this:

```
root@mc-escuela-larrea:~# iwinfo wlan1-mesh scan
Cell 01 - Address: A8:40:41:1C:86:6E
ESSID: "LiMe"
Mode: Mesh Point Channel: 48
Signal: -37 dBm Quality: 70/70
Encryption: none

Cell 02 - Address: A8:40:41:1C:85:25
ESSID: "LiMe"
Mode: Mesh Point Channel: 48
Signal: -86 dBm Quality: 24/70
Encryption: none
```

Where only the mac address can help you to identify each node.
Most of the time you don't have the mac address at hand.
Adding the hostname to the beacon would be very helpful in this case.

I have no clue on how to do this.

der-nele, 2021-04-23:
i like to have this too! in future update :) 
if possible?
is this dependent on the mesh-protocol? 
in my case bmx7 and batman

ilario, 2021-04-23:
@cl4u2 is the right person to help with this issue :)

ilario, 2021-04-24:
Actually, all the needed information is in one of his repositories' readme: https://github.com/cl4u2/ieswescan

spiccinini, 2021-04-24:
Thanks Ilario! If I am not mistaken all these approaches use the "normal AP" beacons and not the 802.11s beacons. So for example the integration is through hotapd that is not used on 802.11s (at last in our setup). I dont know if mixing both kind of beacons is possible and if it is a good idea, nor if it is possible to add more information to the mesh beacons. Adding info to the mesh beacons I believe that it is possible in theory, but I dont know if the kernel is exposing an API to do so (I am not experienced in netlink sockets that, I  think but may be wrong, is the API for this)

der-nele, 2021-04-25:
ok i think what i wont is more a question for bmx7. 
that bmx7 is sharing the channel information in the mesh. 
with router use with channel. for more easy channel tuning.

ilario, 2021-04-25:
> ok i think what i wont is more a question for bmx7.
> that bmx7 is sharing the channel information in the mesh.
> with router use with channel. for more easy channel tuning.

Hey @der-nele can you open a new issue specific for that?
Thanks!

der-nele, 2021-04-26:
yes i put it hier
https://github.com/bmx-routing/bmx7/issues/59
or should it be in lime-packages?