libremesh/lime-packages #886: Pirania Roadmap

spiccinini, 2021-07-04:
As a first step towards a public Pirania roadmap I am adding some points. We should expand / correct / improve this and maybe open specific issues:

TODO
- [ ]  Expose the voucher creation with duration functionality
- [ ]  Have a "emergency" function to force pirania to show for everyone
- [ ]  Fix redirect to the "original site", now goes to for example /cgi-bin/pirania/http%3A%2F%2Fprueba.com%2F
- [ ]  Do not re-add removed vouchers from the shared-state db (see removed.txt).

spiccinini, 2021-09-03:
I am working in #893

luandro, 2024-01-30:
@spiccinini me and @hiurequeiroz are reviewing some Pirania issues and we found this one.

> Be able to enable the portal only in one interface, for e.g. to be used with the community wireless interface (ap interface) without password leaving the apname interface with password

Has been resolved no?

henmohr, 2024-09-26:
Hey! I was working on translatin captive-portal script from iptables to nftables. Opened a PR [here](https://github.com/libremesh/lime-packages/pull/1133). If you could review would be  very nice.