libremesh/lime-packages #887: derive domain from ap_ssid

altergui, 2021-07-05:
right now, `domain` defaults to `thisnode.info`.
if this setting is not properly customized (made unique), and two different clouds (different ap_ssid) are connected to each other, they will have a conceptual conflict. they will both consider themselves authoritative for the same domain (`thisnode.info`) and will have no way to resolve each other hostnames.

my proposal is to derive `domain` from `ap_ssid` by default (i.e. it can always be customized, the same way ip addresses default to something derived from `ap_ssid` but can be hardcoded to something else)

examples:
 * ap_ssid="La Bolsa Libre"
   * domain is set to "la-bolsa-libre.mesh"

 * ap_ssid="\\\ my ^FUNNY^ network, wohoo 2021!! ///"
   * domain is set to "my-funny-network-wohoo-2021.mesh"

so:
* everything non-alphanumeric is squashed down to a single '-'
* '-' at the beginning and the end are stripped
* `.mesh` is appended
* everything is made lowercase (just for human clarity, since DNS is case insensitive anyway)

notably, this proposal will break current usage of browsing to `somenode.thisnode.info` instead of to `somenode/` or `http://somenode`
i personally haven't seen people doing that in the wild, but there's probably someone out there doing such a thing. it happened with `.lan`
https://github.com/libremesh/lime-packages/issues/540#issuecomment-528518075

`thisnode.info` will still work, of course, for accessing the nearest node, as always.

altergui, 2021-07-05:
just to give more context, this proposal is a prerequisite to https://github.com/libremesh/lime-packages/issues/133
that issue is more epic in terms of implementation, but probably no-one will oppose merging such a wonderful feature.

this present issue #887 is trivial to code, but i have the feeling it will instead demand more time to talk about it here.

ilario, 2021-07-06:
Two small things:

"thisnode.info" is also hardcoded into anygw here:
https://github.com/libremesh/lime-packages/blob/92a98fb6f9f9796bcdf26513efa8f5290ea34b56/packages/lime-proto-anygw/files/usr/lib/lua/lime/proto/anygw.lua#L14

So it will maybe still work (even when changing the domain name) for some of its features (I did not verify which ones).

When using .mesh we hit again #540