libremesh/lime-packages #897: limeWeb has error :error on line 226 at column 1: Extra content at the end of the document

ggsnpdsn, 2021-09-23:
Use http://10.163.166.176/cgi-bin/luci/lime/essentials to access router, an error occurs:
This page contains the following errors:
error on line 226 at column 1: Extra content at the end of the document
Below is a rendering of the page up to the first error.

/etc/config/lime:
config lime 'wifi'
        option ap_ssid 'ap_ssid'

config lime 'network'
        option main_ipv4_address '10.%N1.0.0/16'
        option main_ipv6_address 'fd%N1:%N2%N3:%N4%N5::/64'

config lime 'system'
        option hostname 'LiMe-%M4%M5%M6'

config libremap 'settings'

config libremap 'location'

config libremap 'contact'

ilario, 2021-09-24:
lime-web has abandoned since long time, in favour of lime-app.
Can you try if lime-app works for your needs?
Nevertheless, I think there are people out there still using lime-web.
Any help for solving this issue is welcome!

ggsnpdsn, 2021-09-24:
Sometimes, I want to modify the channel or transmit power and other parameters, but the lime-web cannot modify it.Is it possible to add some configuration parameters in lime-app, such as channel, power, encryption password, distance and other basic parameters

ilario, 2021-09-24:
I agree (see https://github.com/libremesh/lime-packages/issues/327).
We need @germanferrero @spiccinini @patacon13 @nicopace opinion here.

ilario, 2021-10-10:
TL;DR: the fix would be to rewrite a lot of stuff

I had a look at what is failing:
this line
https://github.com/libremesh/lime-packages/blob/25c6ebeb027a9e85cdae3c8004b1d4a0bee48df6/packages/lime-webui/luasrc/model/essentials.lua#L90
tries to get the distance parameter that nowadays has been splitted in distance_2ghz and distance_5ghz.
But even replacing the `distance` by one of the other two valid values, the page stays blank (but the error is not there anymore).
In the source code of the `essentials` page, there are plenty of references to LibreMap, which is defunct since long time. So it looks hard to maintain now and would require rewriting a lot of the lime-webui code or at least of the `essentials` page. I would expect no effort in this direction from the core devs, but I'm sure that any contribution fixing things will be welcome anyway.

Even if the error is confirmed, I suggest to close this issue, unless anyone is willing to work on it.
Indeed, we could even remove the lime-webui package from this repository.

ilario, 2024-11-08:
Noone volunteered to fix this in the last 3 years. Closing. @ggsnpdsn feel free to re-open if you think there is an actual chance to get this fixed.

closed 2024-11-08 by ilario