libremesh/lime-packages #909: safe-reboot can be triggered by unauthenticated user

germanferrero, 2022-01-11:
See usr/libexec/rpcd/lime-utils, there is a method for calling safe-reboot. This allow anyone reaching the router ip to reboot it.

ilario, 2022-01-11:
I had no idea, verrrry interesting!!!!
I don't know how to use this, so I managed to get only to this point:
```
$ curl --location --request POST 'http://thisnode.info/ubus' \
--header 'Content-Type: application/json' \
--data-raw '{
    "jsonrpc": "2.0",
    "id": 1,
    "method": "call",
    "params": [
        "00000000000000000000000000000000",
        "lime-utils",
        "safe_reboot",
        {
            "action": "status"
        }
    ]
}'
```

Resulting in

```
{"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}
```

And [reading here](https://openwrt.org/docs/techref/ubus#actually_making_calls) seems that I should get an ID from a previous login step, so it does not look like completely open to non-users to me, but I really never touched this stuff before.
Thanks for finding this interesting issue!

ilario, 2022-12-25:
@selankon @spiccinini @germanferrero can anyone give instructions on how to trigger this? Or can you tell if this is already safe?

ilario, 2023-02-28:
Asked to Ger via email and got details, still to be tested:

> Fíjate de entrar a limeapp con el navegador y la pestaña de network abierta.
> Verás un "ajax" request de login inicial que esto crea un session id para el usuario lime-app.
> 
> Incluyendo ese session id en el request de safe-reboot ya podrás reproducirlo!

selankon, 2023-03-22:
Is there any reason to unprotect this method? I mean, a use case where an admin that can't reach the ip tells to a random person without knowledge to access and reboot? Or is a bug at all?

Either, I cannot find how `lime-utils-admin` methods are protected... I suspect that could be enough to move the `safe_reboot` functions to lime-utils-admin files.

I will try ASAP

ilario, 2023-03-22:
The issue has been opened by @germanferrero, whho implemented a lot of this stuff (I believe), so I would bet it is a bug, not a feature.

selankon, 2023-03-27:
FTR, to get session just do:

```bash
curl -X POST http://thisnode.info/ubus/ -d '{"id":1,"jsonrpc":"2.0","method":"call","params":["00000000000000000000000000000000","session","login",{"username":"lime-app","password":"generic","timeout":5000}]}'
```

Then get the session id, using `jq` you can pipe the result with something like: 

```bash
jq '.result[1].ubus_rpc_session'
```

selankon, 2023-03-27:
This should fix it 

https://github.com/libremesh/lime-packages/pull/989

ilario, 2023-03-27:
Ah hem, I don't know how to check that... Maybe @altergui @nicopace @G10h4ck @spiccinini ?

selankon, 2023-03-28:
Ok looking for it, to get root access: 

```
curl -X POST http://thisnode.info/ubus/ -d '{"id":1,"jsonrpc":"2.0","method":"call","params":["00000000000000000000000000000000","session","login",{"username":"root","password":"<PASSWORD_HERE>","timeout":5000}]}' | jq '.result[1].ubus_rpc_session'
```

So you get a root rpc session. And then just call `safe_reboot` the same way as before with the new rpc session. 

.... And not works! I'm checking it

selankon, 2023-03-28:
Oh ok, I forgot to update the curl from `lime-utils` to `lime-utils-admin`. Now, for the root session works and for the user session return a:

```
{"jsonrpc":"2.0","id":1,"error":{"code":-32002,"message":"Access denied"}}%                                                                                              
```

Here the updated call:

```bash
curl --location --request POST 'http://10.219.123.10/ubus' \
--header 'Content-Type: application/json' \                           
--data-raw '{
    "jsonrpc": "2.0",
    "id": 1,
    "method": "call",
    "params": [
        "cf25bff8ac0281361cb298bbfbfd0364",
        "lime-utils-admin",
        "safe_reboot",
        {
            "action": "status"
        }
    ]
}'
```

And I added a commit correcting a typo.

closed 2023-04-05 by ilario