libremesh/lime-packages #919: prometheus-node-exporter-* packages not present in lime-feed

germanferrero, 2022-03-07:
The packages:

prometheus-node-exporter-lua-wifi_stations,
prometheus-node-exporter-lua-wifi_stations_extra,
prometheus-node-exporter-lua-wifi_survey,
prometheus-node-exporter-lua-wifi_params,

Are not present in lime-feed master: https://github.com/libremesh/lime-feed/tree/gh-pages/master
does anyone have a clue of why could this happen?

germanferrero, 2022-03-07:
@spiccinini @ilario @altergui

ilario, 2022-03-07:
I dug a bit in this and I could not understand.
Anyway those packages should not be on LibreMesh repositories: they should be moved to the OpenWrt ones, see discussion on #871

Maybe those compiled packages are not in bin/packages/x86_64/libremesh/ that is the exported directory as indicated here but for some reason are in bin/packages/x86_64/packages/? Maybe because they get selected as a dependency before being called one by one?
https://github.com/libremesh/lime-packages/blob/72129177d0eb692d0741f0ecec4ead498fb8365b/.github/workflows/build.yml#L38

The Docker file compiling the packages is this:
https://github.com/openwrt/gh-action-sdk/blob/c428502eba9acf6a84c311e4a5632316912c8ed7/entrypoint.sh#L32-L39

And all seems correct. Even checking the build log the prometheus packages got compiled...
https://github.com/libremesh/lime-packages/runs/5095881117?check_suite_focus=true#step:3:2037

germanferrero, 2022-03-08:
Thank you so much @ilario for trying to solve it. Maybe aparcar can give us some light yes

ilario, 2022-03-08:
What you could do is to compile locally and check where the compiled package is.

aparcar, 2022-03-09:
First of, why are those packages not part of upstream?

altergui, 2022-03-11:
these packages were developed as part of a project 2020-2021, which apparently are useful outside of that scope. i'm happy about that! but i'm not maintaining these packages anymore, so if anyone wants to push them upstream, go ahead

closed 2022-03-17 by spiccinini