libremesh/lime-packages #927: ubus-lime-utils file conflict when compiling with OpenWrt 22.03

ilario, 2022-07-14:
There's a file conflict when compiling on top of OpenWrt 22.03.0-rc4 between:
https://github.com/libremesh/lime-packages/blob/400ac95448a1f1d5a83496488e40c302cde766f0/packages/ubus-lime-utils/files/etc/udhcpc.user
and:
https://github.com/openwrt/openwrt/blob/130118f7aa3191a2afd244d49e3057d4c5688676/package/network/config/netifd/files/etc/udhcpc.user

This is what the error looks like:

```
Collected errors:
 * check_data_file_clashes: Package ubus-lime-utils wants to install file /home/ilario/software/openwrt/build_dir/target-mips_24kc_musl/root-ath79/etc/udhcpc.user
	But that file is already provided by package  * netifd
 * opkg_install_cmd: Cannot install package lime-app.
 * check_data_file_clashes: Package ubus-lime-utils wants to install file /home/ilario/software/openwrt/build_dir/target-mips_24kc_musl/root-ath79/etc/udhcpc.user
	But that file is already provided by package  * netifd
 * opkg_install_cmd: Cannot install package ubus-lime-utils.
make[2]: *** [package/Makefile:70: package/install] Error 255
```

spiccinini, 2022-07-14:
Ok! So we should install the file as part of the install step instead?

ilario, 2022-07-15:
I think that the solution that has been used more frequebtly in this repository is the creation of a file in /etc/uci-defaults that will add the needed content at the first boot. Like this: https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/etc/uci-defaults/92_add-lime-repos

G10h4ck, 2022-11-28:
If possible it is always better to do this kind of stuff at compile/install time, it is better on the logical side and we get the benefit of squashfs compression which won't happen doing it at runtime (uci-default) which should be considered a last resort when it is not possible to do things at _cooking_ time.

About this particular issue I wonder if the `udhcpc` doesn't have something like `/etc/udhcpc.user.d/` where additional scripts may be placed, especially now that the file is already provided/used by the package itself

ilario, 2022-11-29:
You are right! The /etc/udhcpc.user.d/ directory exists. I am going to use that instead and open another PR.

ilario, 2022-11-29:
On a side note (I am not yet going to open an issue on this as I am not sure it is a bug):
this file calls the watchping, but the watchping is NOT a dependency of ubus-lime-utils, so this results in an error.
Is this right or there should be a dependency there? @G10h4ck @spiccinini @germanferrero

PR #950 (2023-03-27): ubus-lime-utils place scripts in /etc/udhcpc.user.d/
Fix #927 
Still to be tested. Specifically, it is not obvious that it works on OpenWrt 19.07, while it should work on 22.03 as that directory is already present there.