libremesh/lime-packages #932: Distance setting problem and "auto" setting

ilario, 2022-09-20:
Talking to community members seems that is common the experience of trying LibreMesh over long links and discard it as it "does not work". This is due to the default "distance" setting on the 5 GHz link (1 km):
https://github.com/libremesh/lime-packages/blob/78162b4d773c9e8d01f63801ce3464ea84ac89b6/packages/lime-system/files/etc/config/lime-defaults#L73

There is no "right" value (in my opinion, to have it too large is better than having it too small), and surely is of foremost importance to recommend in the documentation to change it.

Asking around at BattleMeshV14 we just realized that there is an "auto" setting for distance in ath9k driver.
The Aredn people set it and unset it a few times in their firmware https://github.com/aredn/aredn_ar71xx/search?q=distance&type=issues so seems like it had some problems, at least in the past.

I think we could recommend the users to select the
`CONFIG_PACKAGE_ATH_DYNACK=y` (in Menuconfig it is in Kernel modules > Wireless drivers > kmod-ath > Enable Dynack support) and set "auto" or we could even make it default with a fallback value.

Something like: if the distance value is negative (e.g. `-1000` but I would push for `-5000`) and have lime-config checking if Dynack is available (how??????), otherwise just apply the distance number. If the distance is positive, just use that value without checking.

ilario, 2022-09-21:
Thanks to @itay-sho we realized that in order to know if the DynAck (auto distance) is available, we can check for the presence of the `/sys/kernel/debug/ieee80211/phy*/ath9k/ack_to` file that gets created here https://github.com/torvalds/linux/blob/fc02cb2b37fe2cbf1d3334b9f0f0eab9431766c4/drivers/net/wireless/ath/ath9k/debug.c#L1493
This file will also contain the distance, for example now it contains "600 A".
The other option is to run the iw phy phy0 set distance auto (phy0 or phy1) and look at the exit code. If it is 0 it is supported, otherwise it should be 134 (observed on a TP-Link WDR4300 with OpenWrt 22.03.0 with no DynAck support).

spiccinini, 2022-09-24:
Aside from enabling auto in ath9k supported boards, we should increase the default of 1km to at least 5km?

ilario, 2022-10-02:
> we should increase the default of 1km to at least 5km?

Yes, I also think so. Better a poorly performing connection when the link is short than a non-working connection when the link large.

> Something like: if the distance value is negative (e.g. -1000 but I would push for -5000) and have lime-config checking if Dynack is available (how??????), otherwise just apply the distance number. If the distance is positive, just use that value without checking.

Changed my mind, much better to have a true/false option named something like `distance_use_auto_if_available`.

ilario, 2022-10-02:
The documentation on https://github.com/openwrt/openwrt/blob/40c2cd8bd0f7e7b6553d47c11dacb48acc533863/package/kernel/mac80211/ath.mk#L112-L121
says:

> most remote station detected

the "station" in the phrase could refer to AP-sta connections only... Does anyone know if it works only for APs??
What happens if we activate this on a radio with only mesh? Or on a radio with AP+AP+mesh?

It is implemented here:
https://github.com/torvalds/linux/blob/master/drivers/net/wireless/ath/ath9k/dynack.c

spiccinini, 2022-10-03:
Yes, distance configuration affects AP and/or mesh. We have been using the `auto` config for LibreRouterOs as the radios are ath9k with the dynack enabled with `CONFIG_PACKAGE_ATH_DYNACK=y`

ilario, 2022-10-03:
> Yes, distance configuration affects AP and/or mesh.

Affects, sure.

But what about the mechanism for deciding the good distance value?
Is the kernel going to adapt the distance based on the farthest *mesh peer* or does this work only with the farthest *station*?
Did you try to use it on a long ieee802.11s link? (I simply cannot test this, as my flat is too small for that XD)

spiccinini, 2022-10-03:
Yes, we tested and is used in production in some networks with long mesh links. In most places of the kernel code a station is just any "wifi peer" you can communicate, it does not require to be a client. Yes it is confusing.

ilario, 2022-10-04:
@spiccinini, this is a very good news!!
I think we should implement the `distance_use_auto_if_available` and set it to true by default!

ilario, 2024-12-24:
Reporting this just as a curiosity, I don't think we should do it: LibreRouterOS has [auto distance enabled by default](https://gitlab.com/librerouter/librerouteros/-/commit/d108e173b0bd2f26f9ff82684632c9d2c3eda367). I suppose they can do that as they only support devices with ath9k (which supports auto distance).