libremesh/lime-packages #933: Add .test to domains for resolving host names

ilario, 2022-10-02:
Disclaimer: this is not related with the thisnode.info address for connecting to the nearest node which is here:
https://github.com/libremesh/lime-packages/blob/c23828c1e8aa055f143375c4d89352e8ff1c4283/packages/lime-proto-anygw/files/usr/lib/lua/lime/proto/anygw.lua#L14

***

Once upon a time we had the `.lan` domain, that didn't work on Google Chrome browser #540, so we replaced it with `.thisnode.info` #541 so that now the connected hosts can be resolved as `laptop-mario.thisnode.info` (which I think that nobody uses as it is not much known) or typing `http://laptop-mario` or `laptop-mario/` (both unusual and weird for the average user).
This is in the `domain` field of the `/etc/config/lime-*` configuration files 
https://github.com/libremesh/lime-packages/blob/78162b4d773c9e8d01f63801ce3464ea84ac89b6/packages/lime-system/files/etc/config/lime-defaults#L10

which should be replaced by something like `nodes.thenameofmycommunity.org` so that the hosts can be resolved as `laptop-mario.nodes.thenameofmycommunity.org` as documented here:
https://github.com/libremesh/lime-packages/blob/4910e9f7617426454ea261b8c2d4117db5395fda/packages/lime-docs/files/www/docs/lime-example.txt#L20

[Looking on the internet](https://webdevstudios.com/2017/12/12/google-chrome-63/) seems that the `.test` domain should work on every browser.
So, additionally to being able to connect to the hosts in the aforementioned methods, we could also have `laptop-mario.test`

This should be as easy as adding `hostname..".test"` to this list:
https://github.com/libremesh/lime-packages/blob/c23828c1e8aa055f143375c4d89352e8ff1c4283/packages/lime-proto-anygw/files/usr/lib/lua/lime/proto/anygw.lua#L113

Jadhav-Dhanashri, 2026-01-17:
Hello maintainers ,
I'm new to freifunk.net open source contribution & this would be my first issue. I'd like to work on this if it's available & learn from the process. Could you please assign it to me?
Thank you !

ilario, 2026-01-17:
Update: .internal is confirmed to be a good option:
https://www.icann.org/en/board-activities-and-meetings/materials/approved-resolutions-special-meeting-of-the-icann-board-29-07-2024-en#section2.a

more info on:
https://icannwiki.org/.internal

@Jadhav-Dhanashri please feel free to go on with the implementation