libremesh/lime-packages #944: Bridge config changes due to DSA

ilario, 2022-11-06:
Running `lime-config` on a YouHua WR1200JS with OpenWrt 22.03 I got the following error:

```
root@OpenWrt:~# lime-config 
Mac-config: lime-d45f25eb7eac
Clearing wireless config...
Clearing network config...
Disabling odhcpd
Cleaning dnsmasq
Disabling 6relayd...
network.scandevices.owrt_ifname_parser found ifname wlan0-ap
network.scandevices.dev_parser found WiFi device wlan0-ap
network.scandevices.owrt_ifname_parser found ifname wlan0-apname
network.scandevices.dev_parser found WiFi device wlan0-apname
network.scandevices.owrt_ifname_parser found ifname wlan0-mesh
network.scandevices.dev_parser found WiFi device wlan0-mesh
network.scandevices.owrt_ifname_parser found ifname wlan1-ap
network.scandevices.dev_parser found WiFi device wlan1-ap
network.scandevices.owrt_ifname_parser found ifname wlan1-apname
network.scandevices.dev_parser found WiFi device wlan1-apname
network.scandevices.owrt_ifname_parser found ifname wlan1-mesh
network.scandevices.dev_parser found WiFi device wlan1-mesh
/usr/lib/lua/lime/utils.lua:17: bad argument #2 to 'format' (string expected, got nil)
stack traceback:
	/usr/lib/lua/lime/config.lua:214: in function </usr/lib/lua/lime/config.lua:214>
	[C]: in function 'foreach'
	/usr/lib/lua/lime/network.lua:288: in function 'scandevices'
	/usr/lib/lua/lime/network.lua:316: in function </usr/lib/lua/lime/network.lua:307>
	[C]: in function 'xpcall'
	/usr/lib/lua/lime/config.lua:214: in function 'main'
	/usr/bin/lime-config:55: in main chunk
	[C]: ?
Configuring system...
Let uhttpd listen on IPv4/IPv6
Applying generic configs:
 uhttpd_https
 Error on generic config uci_set: uhttpd.main.redirect_https=0
Done applying generic configs.
Copying assets:
Done copying assets.
Running assets on ATCONFIG :
Done running assets.
```

The value for `ifname` here is missing:
https://github.com/libremesh/lime-packages/blob/3bafc4d106e7142cc408a44ea401a0aeec9643bd/packages/lime-system/files/usr/lib/lua/lime/network.lua#L257

This happens due to the changes introduced in OpenWrt 21.02 with the DSA system for configuring bridges, instead of the old swconfig, see the most important changes here
https://openwrt.org/releases/21.02/notes-21.02.0#new_network_configuration_syntax_and_boardjson_change
specifically the point "in `config device` of type `bridge`, `ifname` has been renamed to `ports`".

Additionally, its content changed from a string containing the ports to a table, so that a change in the code is needed.

Also, the parsing in `dev_parser` needs to be adapted to the new naming like `lan1` in these lines:
https://github.com/libremesh/lime-packages/blob/3bafc4d106e7142cc408a44ea401a0aeec9643bd/packages/lime-system/files/usr/lib/lua/lime/network.lua#L224

ilario, 2022-11-06:
Do we need to keep compatibility with devices not supported by DSA?
Most targets have already been migrated, see [here](https://openwrt.org/releases/21.02/notes-21.02.0#initial_dsa_support) and [here](https://openwrt.org/releases/22.03/notes-22.03.0#more_targets_converted_to_dsa).

If we want to keep retrocompatibility with OpenWrt 19.07, we can check if DSA is in place [with this lines](https://forum.openwrt.org/t/how-to-detect-dsa/111868/4).

spiccinini, 2022-11-09:
I believe that we can first work in supporting 22.03 and then check if adding backwards compatibility is needed. What we will have to do is to work in some migration script so an upgrade from 19.07 is viable. Some work is being done here https://github.com/openwrt/openwrt/pull/10796

ilario, 2022-11-22:
I just tried with OpenWrt 19.07 and I got the same error but for another interface.

```
network.scandevices.owrt_ifname_parser found ifname wlan0-ap
network.scandevices.dev_parser found WiFi device wlan0-ap
network.scandevices.owrt_ifname_parser found ifname wlan0-apname
network.scandevices.dev_parser found WiFi device wlan0-apname
network.scandevices.owrt_ifname_parser found ifname wlan0-mesh
network.scandevices.dev_parser found WiFi device wlan0-mesh
network.scandevices.owrt_ifname_parser found ifname wlan1-ap
network.scandevices.dev_parser found WiFi device wlan1-ap
network.scandevices.owrt_ifname_parser found ifname wlan1-apname
network.scandevices.dev_parser found WiFi device wlan1-apname
network.scandevices.owrt_ifname_parser found ifname wlan1-mesh
network.scandevices.dev_parser found WiFi device wlan1-mesh
network.scandevices.owrt_ifname_parser found ifname lo
network.scandevices.owrt_ifname_parser found ifname eth0.1
network.scandevices.dev_parser found vlan device eth0.1 and marking eth0 as nobridge
network.scandevices.owrt_ifname_parser found ifname eth0.2
network.scandevices.dev_parser found vlan device eth0.2 and marking eth0 as nobridge
network.scandevices.owrt_ifname_parser found ifname eth0.2
network.scandevices.dev_parser found vlan device eth0.2 and marking eth0 as nobridge
/usr/lib/lua/lime/utils.lua:17: bad argument #2 to 'format' (string expected, got nil)
stack traceback:
	/usr/lib/lua/lime/config.lua:214: in function </usr/lib/lua/lime/config.lua:214>
	[C]: in function 'foreach'
	/usr/lib/lua/lime/network.lua:288: in function 'scandevices'
	/usr/lib/lua/lime/network.lua:316: in function </usr/lib/lua/lime/network.lua:307>
	[C]: in function 'xpcall'
	/usr/lib/lua/lime/config.lua:214: in function 'main'
	/usr/bin/lime-config:55: in main chunk
	[C]: ?
```

And it was caused by this entry in `/etc/config/network`:

```
config device 'lan_eth0_1_dev'
	option name 'eth0.1'
	option macaddr 'd4:5f:25:eb:7e:ac'
```

This is very weird. I am going to try to see why this happens when installing packages one by one but not when flashing the whole thing together.

ilario, 2022-11-22:
PS it happens because this entry in `/etc/config/network` does not have the `ifname` field here:

ilario, 2022-11-22:
Waaaait, this whole thing has not been caused by DSA but by this commit: 3bafc4d106e7142cc408a44ea401a0aeec9643bd merged in #900 on the 15th of September.

Specifically, if something has a `nil` value, it gets skipped here (and this was happening also before that commit):

https://github.com/libremesh/lime-packages/blob/3bafc4d106e7142cc408a44ea401a0aeec9643bd/packages/lime-system/files/usr/lib/lua/lime/network.lua#L212-L216

what changed in that commit is that before skipping the `nil` interface, a log message is printed and the `nil` value causes an error:

https://github.com/libremesh/lime-packages/blob/3bafc4d106e7142cc408a44ea401a0aeec9643bd/packages/lime-system/files/usr/lib/lua/lime/network.lua#L255-L263

So this is NOT related to DSA.

closed 2022-11-27 by G10h4ck

ilario, 2022-12-10:
> What we will have to do is to work in some migration script so an upgrade from 19.07 is viable.

Thinking about the migration: `sysupgrade` complains when doing the upgrade and asks for forcing the upgrade with `-F` and this usually means that `-n` is needed. If we do the migration script, then we should upgrade with `-F` but without `-n`?

PR #948 (2022-11-27): network.lua use an alternate string if ifname is not found by owrt_device_parser
Fix #944