libremesh/lime-packages #949: unique_append should check if /etc/crontabs/root file exist before appending

ilario, 2022-11-28:
The onliner function used often in the code for adding a line to the root crontab does not check for its existence before looking into it with grep. For example here:

https://github.com/libremesh/lime-packages/blob/5398a91f50abeb5988c56de41aec7bf20da361f8/packages/shared-state/files/etc/uci-defaults/shared-state-publishers-cron#L3-L10

This causes an error on the very first execution of such function:

```
Configuring shared-state.
grep: /etc/crontabs/root: No such file or directory
```

Nevertheless, the content gets added to the crontab anyway, so everything should work even when this error message appears.

ilario, 2024-11-08:
Currently two somutions have been proposed:

Just editing all the files:
https://github.com/libremesh/lime-packages/pull/980/commits/ff8fd58d45c0b3fcb7119b2eceaad256446e41d5

Creating a function to be used until when unique_append gets included in OpenWrt's built-in list of functions (but we did not propose that to OpenWrt yet):
https://github.com/libremesh/lime-packages/pull/988

I like the second option as we should always try to push useful code to OpenWrt.

Jadhav-Dhanashri, 2026-01-12:
Hello maintainers,
I'm new to  freifunk.net open source contributions & excited to get started. I would like to work on this issue, & I'll follow the contribution guidelines carefully. Could you please assign it to me if it's available?
Thank you!