libremesh/lime-packages #951: Adapting LibreMesh to OpenWrt 22.03

ilario, 2022-11-30:
Up to now, LibreMesh works with OpenWrt 19.07 and 18.06.
The tests on OpenWrt 21.02 and 22.03 were not very successful yet.
Some comments from the mailing list can be read here: (mailing list server is down, adding once it gets up again)

Very likely, we will have to adapt to support DSA way to configuring switches rather than the traditional swconfig, see the last part of this comment https://github.com/libremesh/lime-packages/issues/944#issue-1437483339

Here I report the outcome of lime-config configurations when running them on OpenWrt 19.07 and on OpenWrt 22.03.
At a first sight, some differences in /etc/config/network can be spotted and are likely the source of our problems.

**On OpenWrt 19.07**

Output of lime-report on clean OpenWrt 19.07:

[OpenWrt19-plain.txt](https://github.com/libremesh/lime-packages/files/10121271/OpenWrt19-plain.txt)

Console output during the lime-system installation:

[install-lime-system.txt](https://github.com/libremesh/lime-packages/files/10121277/install-lime-system.txt)

Output of lime-report on OpenWrt 19.07 after installing lime-system and rebooting:

[OpenWrt19-with_lime-system.txt](https://github.com/libremesh/lime-packages/files/10121282/OpenWrt19-with_lime-system.txt)

**On OpenWrt 22.03**

Output of lime-report on clean OpenWrt 22.03:

[OpenWrt22-plain.txt](https://github.com/libremesh/lime-packages/files/10121287/OpenWrt22-plain.txt)

Console output during the lime-system installation:

[install-lime-system.txt](https://github.com/libremesh/lime-packages/files/10121286/install-lime-system.txt)

Output of lime-report on OpenWrt 22.03 after installing lime-system and rebooting:

[OpenWrt22-with_lime-system.txt](https://github.com/libremesh/lime-packages/files/10121289/OpenWrt22-with_lime-system.txt)

ilario, 2022-12-12:
Here go more outputs from lime-report when installing more packages:

**On OpenWrt 19.07**

Output of lime-report on OpenWrt 19.07 after installing lime-system and lime-proto-anygw:

[OpenWrt19-with_lime-system_lime-proto-anygw.txt](https://github.com/libremesh/lime-packages/files/10206154/OpenWrt19-with_lime-system_lime-proto-anygw.txt)

Output of lime-report on OpenWrt 19.07 after installing lime-system and lime-proto-anygw and lime-proto-batadv and lime-proto-babeld:

[OpenWrt19-with_lime-system_lime-proto-anygw-batadv-babeld.txt](https://github.com/libremesh/lime-packages/files/10206156/OpenWrt19-with_lime-system_lime-proto-anygw-batadv-babeld.txt)

**On OpenWrt 22.03**

Output of lime-report on OpenWrt 22.03 after installing lime-system and lime-proto-anygw:

[OpenWrt22-with_lime-system_lime-proto-anygw.txt](https://github.com/libremesh/lime-packages/files/10206157/OpenWrt22-with_lime-system_lime-proto-anygw.txt)

Output of lime-report on OpenWrt 22.03 after installing lime-system and lime-proto-anygw and lime-proto-batadv and lime-proto-babeld:

[OpenWrt22-with_lime-system_lime-proto-anygw-batadv-babeld.txt](https://github.com/libremesh/lime-packages/files/10206159/OpenWrt22-with_lime-system_lime-proto-anygw-batadv-babeld.txt)

ilario, 2023-02-15:
An update from the mailing list https://lists.autistici.org/message/20230215.104531.08767d53.en.html :

>> support for OpenWrt 22.03, updates about:
>> https://github.com/libremesh/lime-packages/pull/959  ?
>
> When I wrote that pull request, I thought that the new configuration format was due to the swconfig-DSA migration.
> But instead it is due to the OpenWrt 19.07 vs >19.07 versions, so I have to rewrite the pull request.
> 
> Considering this small issue (would be easy to rewrite that) and the big issue of the new firewall used in OpenWrt 22.03 (iptables and ebtables have been abandoned in favor of nftables) we had a short discussion in the Element chat and in the linked pull request about supporting only OpenWrt 22.03 in the lime-packages' "master" branch. So we would create a "target_OpenWrt_19.07" branch for the extended support to OpenWrt 19.07 and a tag with a minor release "2020.2" which can be considered as an updated version of the 2020.1 LibreMesh release "Expansive Emancipation".
> Can you discuss also this in the meeting?
> 
> Other relevant issues for using OpenWrt 22.03 are:
> https://github.com/libremesh/lime-packages/pull/971 (Dnsmasq now can access only specific folders)
> https://github.com/libremesh/lime-packages/pull/950 (we were creating a file that now is provided by an OpenWrt package)
> 
> And the need to migrate all our iptables and ebtables commands and dependencies to nftables (supposedly, nftables commands are backcompatible in OpenWrt with iptables commands, but the dependencies are quite different, and the result is that installing a trivial package like lime-proto-wan pulls a crazy amount of dependencies).

FreifunkUFO, 2023-04-06:
yes, old ticket: nice brunch names would be conventional!
i heard, in leipzig some libremesh nodes are running OpenWrt 21.02 since months.

closed 2024-11-08 by ilario